microsoft/vscode-arduino #1243: Raspberry Pi | cannot select port

CAProjects, 2021-04-23:
I can get the extenssion working on a windows PC, however on a Raspberry Pi 32bit I am unable to select a port. This makes it impossible to upload code to the Arduino board.

I am able to select the board with out issue, its just the port i cannot

davidcooper1, 2022-02-15:
@CAProjects have you checked to see if your Raspberry Pi is able to detect any devices? I've seen a few sites list `dmesg | grep tty` as a way to see any messages relating to serial activity. Working fine on my Raspberry Pi 64bit Lite OS. I think the packages should come pre-installed. If not, the arduino package through `apt-get` should have them.

davidcooper1, 2022-02-15:
Could be related to #1207 and #1181

closed 2023-03-26 by github-actions[bot]