microsoft/vscode-arduino #1250: Unable to select Arduino USB serial port for programming

unix-d00d, 2021-05-05:
Issue Type: <b>Bug</b>

Running on old iMac 2008 with OS X El Capitan (10.11.6). Just updated to April 2021 edition of VS Code. Version info below.

Version: 1.56.0
Commit: cfa2e218100323074ac1948c885448fdf4de2a7f
Date: 2021-05-04T22:06:21.189Z
Electron: 12.0.4
Chrome: 89.0.4389.114
Node.js: 14.16.0
V8: 8.9.255.24-electron.0
OS: Darwin x64 15.6.0

I noticed when I opened two instances of Code with Arduino extension, the app crashed. Upon reopening the app, I was able to set Arduino Board Configuration on the new project. However, when I clicked on the serial port, there is no popup available. The work around was to manually edit the .vscode/arduino.json and add entry:

"port": "/dev/tty.usbmodemFA431"

Restart VS Code to resume.

Extension version: 0.4.2
VS Code version: Code 1.56.0 (cfa2e218100323074ac1948c885448fdf4de2a7f, 2021-05-04T22:06:21.189Z)
OS version: Darwin x64 15.6.0

ltan10, 2021-05-06:
Having same issue of unable to select serial port, did not get a chance to replicate app crash with second vscode instance.
Tempory fix: roll back vscode to version 1.55.2 https://code.visualstudio.com/updates/v1_55

SebiAi, 2021-05-06:
> Having same issue of unable to select serial port, did not get a chance to replicate app crash with second vscode instance.
> Tempory fix: roll back vscode to version 1.55.2 https://code.visualstudio.com/updates/v1_55 (don't forget to disable auto update)

Thank you very much! This worked great! Works again now!

For anybody wondering what version to download:
If your VS Code is installed in `C:\ProgramData\` then you have the `System `version, otherwise if installed in `%localappdata%\Programs\` you have the `User `version.