microsoft/vscode-arduino #1257: Failed to open serial port due to error on VSC `v1.56.0`

rrakso, 2021-05-07:
Issue Type: <b>Bug</b>

Failed to open serial port /dev/ttyUSB0 due to error: + Error: The module '/home/oskar/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/usb-detection/build/Release/detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 85. This version of Node.js requires
NODE_MODULE_VERSION 87. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Extension version: 0.4.2
VS Code version: Code 1.56.0 (cfa2e218100323074ac1948c885448fdf4de2a7f, 2021-05-04T21:58:14.757Z)
OS version: Linux x64 5.8.0-50-generic snap

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-8350U CPU @ 1.70GHz (8 x 3600)|
|GPU Status|2d_canvas: unavailable_software<br>gpu_compositing: disabled_software<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>opengl: disabled_off<br>rasterization: disabled_software<br>skia_renderer: enabled_on<br>video_decode: disabled_software<br>vulkan: disabled_off<br>webgl: unavailable_software<br>webgl2: unavailable_software|
|Load (avg)|2, 2, 2|
|Memory (System)|7.63GB (0.31GB free)|
|Process Argv|. --no-sandbox --crash-reporter-id a77eba79-e7c6-4e2d-be00-7fbfd8d4d24e|
|Screen Reader|no|
|VM|0%|
|DESKTOP_SESSION|ubuntu|
|XDG_CURRENT_DESKTOP|Unity|
|XDG_SESSION_DESKTOP|ubuntu|
|XDG_SESSION_TYPE|x11|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368cf:30146710
vsreu685:30147344
python383cf:30185419
pythonvspyt700cf:30270857
pythonvspyt602cf:30294773
vspor879:30202332
vspor708:30202333
vspor363:30204092
vstes627cf:30244335
pythonvsnew554cf:30291488
pythontb:30283811
pythonvspyt551cf:30291415
vspre833:30267464
pythonptprofiler:30281270
vshan820:30294714
pythondataviewer:30285071
vscus158cf:30286554

```

</details>

vvinogradsky, 2021-05-07:
Same issue here (running Windows 10)
After removing and reinstalling the Arduino extension I now get "detection.node is not a valid Win32 application". Opening the file in a HEX viewer shows ELF signature, so it looks like somehow I got an executable for the wrong OS.

mnsuk, 2021-05-07:
Same issue here:
windows 10.
VSC v1.56.0
node v16.1.0
vscode-arduino  v0.4.2

**detection.node is not a valid Win32 application.**

`> file detection.node
detection.node: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=97745d38cf6d793b9fbccd3dbbc5dae6ad88c705, stripped`

jatu-studiobox, 2021-05-07:
Same issue
windows 10
VSC v1.56.0
vscode-arduino v0.4.2

After run 'Arduino: Open Serial Monitor' at command palette. The error occured.

Failed to open serial port COM3 due to error: + Error: The module '\\?\c:\Users\TERU\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.2\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 85. This version of Node.js requires
NODE_MODULE_VERSION 87. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

But it can remain upload code to device.

mMerlin, 2021-05-07:
Here too.

* Fedora 33
* Visual Studio Code Version: 1.56.0
* Commit: cfa2e218100323074ac1948c885448fdf4de2a7f
* Date: 2021-05-04T21:58:14.757Z
* Electron: 12.0.4
* Chrome: 89.0.4389.114
* Node.js: 14.16.0
* V8: 8.9.255.24-electron.0
* OS: Linux x64 5.11.17-200.fc33.x86_64
* vscode-arduino: v0.4.2

From a similar problem [reported last October](https://github.com/microsoft/vscode/issues/106500#issuecomment-691426546), the solution seems to be to get the `usb-detection` module recompiled for the new version (of Electron?). Which needs to be done by the vscode-arduino maintainers, then push the changes out so the extension gets updated. A temporary fix MAY be to downgrade vscode to an older version.

Aksuups, 2021-05-07:
I do have the very same problem also. Brief browsing seems to indicate that this is happening quite often.
Like @mMerlin wrote "From a similar problem reported last October".

I'm quite curious though, why my usb-detection was compiled using NODE_MODULE_VERSION 67?
That is 20 versions behind, and way older than others have on their vscode installations. Well,

Version: 1.56.0
Commit: cfa2e218100323074ac1948c885448fdf4de2a7f
Date: 2021-05-04T21:58:14.757Z
Electron: 12.0.4
Chrome: 89.0.4389.114
Node.js: 14.16.0
V8: 8.9.255.24-electron.0
OS: Linux x64 4.18.0-301.1.el8.x86_64

As a workaround I'm using screen for the /dev/ttyUSB0......

mMerlin, 2021-05-08:
> Brief browsing seems to indicate that this is happening quite often.

And is very likely to continue to happen. The extension maintainers likely do not know ahead of time when a vscode update will be pushed that uses a new version. All they can do is react when an update breaks the extension.

How long that takes could vary a lot. If it is a simple matter of changing a number, compile, commit, push, then it can be fast. If changing the version number potentially breaks other things as well, then a full regression test cycle is needed. Which takes longer to start with, and may need additional changes before an update can be sent out.

atcolombini, 2021-05-08:
> 
> 
> Issue Type: Bug
> 
> Failed to open serial port /dev/ttyUSB0 due to error: + Error: The module '/home/oskar/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/usb-detection/build/Release/detection.node'
> was compiled against a different Node.js version using
> NODE_MODULE_VERSION 85. This version of Node.js requires
> NODE_MODULE_VERSION 87. Please try re-compiling or re-installing
> the module (for instance, using `npm rebuild` or `npm install`).
> 
> Extension version: 0.4.2
> VS Code version: Code 1.56.0 (cfa2e218100323074ac1948c885448fdf4de2a7f, 2021-05-04T21:58:14.757Z)
> OS version: Linux x64 5.8.0-50-generic snap
> System Info
> A/B Experiments

In this issue is a list of previous occurrences of this, if you want to link these here:
https://github.com/microsoft/vscode-arduino/issues/1198

membiblio, 2021-05-09:
I had the same issue and found another thread discussing:
https://github.com/microsoft/vscode-arduino/issues/1074

Andrew Trullock provided guidance as follows:
'Not at my computer to cite the exact versions, but the latest Arduino extension works with the previous VSC version from April'

So the fix is to use an older version - here is what worked for me with Andrews help:

Visual Studio Code 1.55.2 from April 21 which can be downloaded here (to assist others):
https://code.visualstudio.com/updates/v1_55

And the Visual Studio Code Extension: Arduino for Visual Studio Code 0.4.2
Which was already installed in the ide as it remembered it even after uninstalling and installing 1.55.2

This combination will work as you expect it to and I can confirm that it does :)

RedaBousbah, 2021-05-10:
Hi, same on Pop!Os (Ubuntu).
After the update, I was unable to change the board from the board manager, nor the serial port, the menu would simply not appear.
I downgraded to 1.52 which was the last one that was working for me.
I am using that since.

matdombrock, 2021-05-10:
Same issue here in Windows 10. Using insider and standard. Both have the same problem described in this thread.

adiazulay, 2021-05-12:
#1262 resolves the issue. v0.4.3 will be ready to download as soon as the build is done.

skelter2014, 2021-05-12:
roll back does work.

Version: 1.55.2 (system setup)  **<-- Rolling back to this version resolved issue.****
Commit: 3c4e3df9e89829dce27b7b5c24508306b151f30d
Date: 2021-04-13T09:35:57.887Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042

closed 2021-05-12 by adiazulay

rrakso, 2021-05-13:
To everyone who comes here :smile: 
Guys from Microsoft has released the version `0.4.3` of arduino extension, just [download it](https://github.com/microsoft/vscode-arduino/releases/download/v0.4.3/vscode-arduino-0.4.3.vsix) and do:

```bash
code --install-extension /path/to/extension.vsix
````

### Happy coding! :technologist: 

#### Reference
- https://github.com/microsoft/vscode-arduino/issues/1198#issuecomment-774746151