microsoft/vscode-arduino #1265: IntelliSense includePath issues for Arduino Nano 33 BLE

aufflick, 2021-05-18:
The set of paths auto-included isn't sufficient for the Arduino Nano 33 BLE. The first missing include file is PinNames.h. I dutifully manually found and added all the right paths to c_cpp_properties.json only to have the file re-written over the top of my changes (otherwise I would have included the list of required paths here) :/

So the real bug is the lack of required paths, but it would also be useful to have a way to manually add additional includePath's that aren't overwritten. (It would also be useful to have a comment indicating that any manual changes will be overwritten, but since this file is json that isn't possible).

```
            "includePath": [
                "/Users/me/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/cores/arduino",
                "/Users/me/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/variants/ARDUINO_NANO33BLE",
                "/Users/me/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/cores/arduino/api/deprecated",
                "/Users/me/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/cores/arduino/api/deprecated-avr-comp",
                "/Users/me/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1",
                "/Users/me/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/arm-none-eabi",
                "/Users/me/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/backward",
                "/Users/me/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/lib/gcc/arm-none-eabi/7.2.1/include",
                "/Users/me/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/lib/gcc/arm-none-eabi/7.2.1/include-fixed",
                "/Users/me/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include"
            ],
            "forcedInclude": [
                "/Users/me/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/cores/arduino/Arduino.h"
            ],

```

FWIW PinNames.h is in `/Users/me/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.1.0/cores/arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_ARDUINO_NANO33BLE/PinNames.h` and then the next missing header is `cmsis.h`.

If someone can point me to a standalone PR that fixes a different board I'm happy to populate a PR with the required include paths for this board.

derpda, 2021-06-29:
It seems just adding the line 
```
"/home/peter/.arduino15/packages/arduino/hardware/mbed_nano/2.2.0/cores/arduino/mbed/**"
```
to `"includePath"` fixes it for me. 
This might be putting too many paths in the search path though...
However, this was enough for me to get IntelliSense going.

codingatmorning, 2022-03-14:
@derpda 
> It seems just adding the line
> 
> ```
> "/home/peter/.arduino15/packages/arduino/hardware/mbed_nano/2.2.0/cores/arduino/mbed/**"
> ```
> 
> to `"includePath"` fixes it for me. This might be putting too many paths in the search path though... However, this was enough for me to get IntelliSense going.

The problem as described is the includePath gets overwritten when building, ect.

codingatmorning, 2022-03-14:
If you want to add your own include paths use forceInclude

holgerurbanek, 2022-04-26:
It would be great, if we would get a option in the .vscode/arduino.json to add additional include-files and include-directories, where they not always gets overwritten.

Additionally: Please enhance the auto-generation for the mbed-platforms and also add the standard location for user-installed-libs.

glektarssza, 2022-05-01:
Interestingly enough if I look at my built output (by specifying an `output` path in my `arduino.json` file) I can see there's a `compile_commands.json` file generated. That has a line in it that specifies to use the lines of a text file (`C:\\Users\\<USERNAME>\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_nano\\3.0.1\\variants\\ARDUINO_NANO33BLE/includes.txt` in my case). That file appears to contain all the include paths needed to compile programs. Adding those (many) lines to my `c_cpp_properties.json` has resolved all intellisense errors for me.

Cycov, 2023-05-16:
I tried @glektarssza and @derpda's solutions, does not work. @codingatmorning to preserve whatever you put there, just set `"arduino.disableIntelliSenseAutoGen": false`