microsoft/vscode-arduino #1274: Auto-configure include paths

Garfius, 2021-05-27:
There shold be a way not to spend so many time finding arduino libraries, and pointing each 3rd party library in the c_cpp_properties.json.

Intellisense should not be an extra work once you install this extenmsion.

Other software does it, this one shall do it too.

Nightenom, 2021-06-15:
You can just type the include clause and then run `verify` or `rebuild IntelliSense Config`, but I agree it's annoying.
I would really like to see if at least all libraries in `sketchbook/libraries` are always included.

mattgodbolt, 2021-07-10:
I've not had this work. I'm a newcomer to this plugin and I had to manually add configuration. The "Rebuild IntellSense Config" didn't seem to work, though the project compiles just fine.

mattgodbolt, 2021-07-10:
I take it back: I found the "Arduino" button in the bottom right (the docs do say this but it wasn't clear"