microsoft/vscode-arduino #1277: unknown register name "r0"C/C++(1118)

vde69, 2021-05-31:
const uint8_t daysInMonth [12] PROGMEM = { 31,28,31,30,31,30,31,31,30,31,30,31 };
uint32_t days = D;
for (uint8_t i = 1; i < M; ++i)
    days += pgm_read_byte(daysInMonth + i - 1);  // <<<<< unknown register name "r0"

-----------------------------------
v0.4.3

vde69, 2021-05-31:
the problem was solved by deleting the c_cpp_properties file.json, and the subsequent manual configuration of the directories.

ttttttom, 2021-09-01:
may i know what you have included in the c_cpp_properties in order to solve your problem?
thanks in advance!

cinderblock, 2022-03-02:
When using inline assembly or global register features of GCC, IntelliSense doesn't know how to handle it. I'm still looking for a way to have it not fail when using AVR registers.

```C++
register uint8_t example asm("r2");
```