microsoft/vscode-arduino #1279: please add "__AVR_ATmega328P__" to c_cpp_properties.json when the selected board is arduino uno

lozziboy, 2021-06-01:
I open this ticket to make a request: add the following define to file c_cpp_properties.json when the selected board is arduino uno
`"__AVR_ATmega328P__"`

this is required to avoid intellisense wrong errors detection with the chip registers name
[image]

EKlatt, 2021-06-03:
Does Not work, because cpp.properties.json will be overwritten from Verifying.

lozziboy, 2021-06-03:
> Does Not work, because cpp.properties.json will be overwritten from Verifying.

That's the obvious reason why I opened this ticket

BelKed, 2021-07-01:
I suggest adding that for all boards... For example: For ATtiny85 should be added `__AVR_ATtiny85__`

tonymitchell, 2021-07-04:
This needs to be fixed in the cocopa package this code uses.  See issue https://github.com/elektronikworkshop/cocopa/issues/7
The problem is the compiler options (such as -mmcu=atmega328p) aren't included in the command that gets parsed to produce the list of defines.

dasfdlinux, 2021-10-06:
> This needs to be fixed in the cocopa package this code uses. See issue [elektronikworkshop/cocopa#7](https://github.com/elektronikworkshop/cocopa/issues/7)

The repo you referenced hasn't been updated in over a year, but has a pull request which you mentioned may fix the issue. Wouldn't it be worth pulling that code into this extension, or forking the repo, applying the pull request, and then closing out this issue?

EverlastEngineering, 2022-05-19:
I'm having this issue too. I have a humble suggestion: could we add `${default}` entries to both `includePath` and `forcedInclude`?

This affords a lot of freedom by allow the end user to specify entries in the settings.json in `C_Cpp.default.includePath` and `C_Cpp.default.forcedInclude` and should afford a very low chance of a regression IMO.

NetworkAndSoftware, 2022-06-24:
A workaround is to first create your project, pick your board and allow the intellisense to generate the .vscode/c_cpp_properties.json and then disable the overwriting of it by setting  "intelliSenseGen": "disable" in .vscode/arduino.json. After this you can just add __AVR_ATmega328P__ or whatever you need for your board to .vscode/c_cpp_properties.json.

It's what @lozziboy already suggested above, but I thought people might like the detailed steps.

al3x-huang, 2022-11-21:
I came back to this issue again after many months and forgot how to do the workaround so I'll leave this more detailed walkthrough of the workaround @NetworkAndSoftware and @lozziboy mentioned here just in case I or anyone else need it in the future:

You can have `c_cpp_properties.json` generated automatically initially by searching and running `Arduino: Rebuild IntelliSense Configuration` in `View->Command Palette` or `cmd+shift+p` for mac. After this, check `.vscode/arduino.json` to ensure a configuration with name `Arduino` has been generated and also make sure C/C++. `.vscode/c_cpp_properties.json` should look something like the snippet below. If you don't get this make sure you don't have `Arduino: Disable Intelli Sense Auto gen` checked/disabled in the extension settings.
```
{
    "version": 4,
    "configurations": [
        {
            "name": "Mac",
            "compilerPath": "/usr/bin/clang",
            ...
        },
        {
            "name": "Arduino",
            ...
        }
```

Make sure you use the configuration we just generated by searching and running `C/C++: Select a Configuration...` in command palette as above or alternatively clicking on the bottom bar as shown in this screenshot
[image]

At this point, most of your intellisense errors should be fixed but if you're using any IO port definitions you'll still get an error. To fix this, add a line to `defines` referencing your specific board under the Arduino C++ configuration. As of writing this edit, I am using an ATTiny85 and the definition is `"__AVR_ATtiny85__"`. You can find a list of these defines at [<avr/io.h>](https://github.com/vancegroup-mirrors/avr-libc/blob/master/avr-libc/include/avr/io.h#L385). So your `.vscode/c_cpp_properties.json` should look something like this:

```
{
    "version": 4,
    "configurations": [
        {
            "name": "Mac",
            "compilerPath": "/usr/bin/clang",
            ...
        },
        {
            "name": "Arduino",
            ...
            "defines": [
                "__AVR_ATtiny85__", <---- or your board's define
                "F_CPU=1000000L",
                "ARDUINO=10819",
                "ARDUINO_attiny",
                ...
                ],
            ...
        }
```

Lastly, add `"intelliSenseGen": "disable"` to `.vscode/arduino.json` to prevent your changes from being overwritten. Note that you need to set this to true if you need to update the generated `c_cpp_properties.json` configuration in the future.

Your intellisense should now work!