microsoft/vscode-arduino #1281: Can't find Arduino IDE

stonez56, 2021-06-07:
It seems I couldn't get this to work. VScode showed the Error message!
For Arduino: Path: I have tried the following and non-of them worked:

- C:\Program Files (x86)\Arduino
- "C:\Program Files (x86)\Arduino"
- C:\\Program Files (x86)\\Arduino
- C:/Program Files (x86)/Arduino
- "C:\Program Files (x86)\Arduino"  

Error message:
[image]

Arduino path settings:
[image]

Settings.json:
[image]

Arduino IDE Installed location:
[image]

stonez56, 2021-06-07:
Never mind... I found what happened.... I need to change both "User & Workspace" for Vscode to find Arduino IDE.

In User:
[image]

In Workspace:
[image]

[Suggestion] Recommend to put a browse button to location the Arduino installed location...or even better to search Arduino installed location!

Thanks

benmcmorran, 2022-01-19:
I'm glad you got this working! Regarding your suggestion, [the extension should already be searching for a system-wide Arduino IDE installation](https://github.com/microsoft/vscode-arduino/blob/cdf1d94a5ccdb8362bb5e5f7caf4cf35a0f358cb/src/common/sys/win32.ts#L9-L29), but it looks like that logic didn't work for you. Can you tell me what you get when you run these commands from an *administrator* command prompt?

1. `reg query HKLM\SOFTWARE\WOW6432Node\Arduino`
2. `where arduino`

closed 2022-02-25 by github-actions[bot]