microsoft/vscode-arduino #1283: Arduino verify/upload is now extremely annoying

iqapps, 2021-06-08:
After one of the latest updates, the verify operation for an Arduino project does not show compiled size of the file to upload, unless the verbose option is set.

But when verbose option is set, then upload creates many many lines of just the hash character during the upload operation and also the verify written code, so the important information scrolls out of view.

Could we please get the output structure back as it was before. Right now its practically useless for production environments.

iqapps, 2021-06-08:
Output sample: **Arduino Extension Verbose mode off, Uploading**

**
Preparing boards...
Verifying...
Uploading...
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
**

This is the other end of the scale. No information whatsoever...

blackpancake, 2021-07-03:
Agree.The compiled size is important for a developer.

timdmackey, 2021-08-08:
Possible duplicate of https://github.com/microsoft/vscode-arduino/issues/1215. The compiled size is a must!

dealmeal, 2021-08-28:
Verifying using Arduino IDE is realy faster than VS Code Extension and you earn the memory usage info. Probably the next IDE will solve our problems.

benmcmorran, 2022-01-13:
#1383 should fix this issue for English locales. The fix is merged and will be available in the next release of the extension. Work for other languages is tracked by #1417