microsoft/vscode-arduino #1284: Arduino upload with compile error switch to serial port view when its open

iqapps, 2021-06-08:
Conditions:
  The serial port open
  The code cannot compile. (ex. missing terminating semicolon)
  Output view is on Serial port
  You press "upload"

The output view changes to Arduino during compile
After the compile finishes the output view changes to serial port anyway.

The expected behavior is to stay in Arduino view and show the errors.

gcampbell-msft, 2023-02-16:
@iqapps We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue or if the issue still persists!

iqapps, 2023-02-16:
Thanks, I'll try it out at next convenience.

closed 2023-03-26 by github-actions[bot]