microsoft/vscode-arduino #1287: Arduino Extension runs 'verify' on VS-Code startup

iqapps, 2021-06-11:
The latest release seem to run Arduino "verify" on its own when I start VS-Code and a Arduino project is loaded automatically.

For my part, it overwrites a the hex file, that maybe other parts of my git branch depends on. So, for my part, its a cause of errors and therefore a time stealer, as the hex file changes compile date and time or simply disappear if the last closed code could not compile.

Please dont make decisions on the community behalf and implement features that are disrupting the workflow and please dont implement new behavior without disclosing information about it in the release notes.

Please remove automatic verify as soon as possible or at lease make it workspace/user configurable.

Daedelos, 2021-06-11:
Agreed. The ability to turn off the auto verify on start/load would be such a time saver. I spent ages looking for a way to turn this off.

RevaDiaz, 2021-07-08:
This, and also turn off auto verify every time the board config change. It's annoying when i change the options in board configuration like 6 options, and verify each time.

magouero, 2021-11-01:
Hello,
Agree also. No need to run verify at startup, nor when something changes. 
Having this as an option or a popup asking "want to verify ?" maybe better.

Any way to prevent it today ?

iqapps, 2021-11-08:
Any news on this - really annoying - undocumented - feature ???

robotdad, 2022-02-07:
I'm not observing this behavior. From my perspective this would be beneficial if it did verify on open as IntelliSense doesn't work until a build (via verify) has run. I totally get that it shouldn't overwrite a previous build though. With CMake there is a difference between generate and a full build, we only need the former for IntelliSense. Seems like something like that would be beneficial here to not invalidate previous build results.

aleun, 2022-03-18:
I've been able to repro this behavior in the latest release. 

The root cause is that in #1183 we added a feature to enable regeneration of the Intellisense configuration on changes by registering a rebuild whenever settings change:

https://github.com/microsoft/vscode-arduino/blob/6d15746db89781560963f316a7bc7be7625d2ad9/src/arduino/arduino.ts#L110-L113

Since the change event handlers fire on initial settings load, this had the side effect of also running analysis on load:

https://github.com/microsoft/vscode-arduino/blob/6d15746db89781560963f316a7bc7be7625d2ad9/src/arduinoActivator.ts#L27-L32

To recover the previous behavior, we could move `await arduinoApp.initialize();` after `await deviceContext.loadContext();` so that doesn't happen. To make it user-definable, we can turn this into a setting like `arduino.analyzeOnOpen` that defaults to `false`, as would be consistent with `cmake.configureOnOpen`. @robotdad @benmcmorran what do you think?

benmcmorran, 2022-03-24:
Thanks for the feedback @iqapps @Daedelos @rev1204 @magouero @BelKed! I've opened PR #1480 that adds two new options to make this configurable. Can you uninstall the Arduino extension, [install the appropriate VSIX from this CI build](https://github.com/microsoft/vscode-arduino/actions/runs/2031489294), and let me know if it meets your needs? These new options can be set in settings.json:

- `arduino.analyzeOnOpen` When true, automatically run analysis when the project is opened. Only works when `arduino.analyzeOnSettingChange` is true.
- `arduino.analyzeOnSettingChange` When true, automatically run analysis when board, configuration, or sketch settings are changed.

Both options default to true so that IntelliSense continues to work out-of-the-box in the default configuration (like robotdad mentioned), but can be set in the user-level settings.json if you want them to be disabled in all your Arduino projects.

rev1204, 2022-03-24:
Thank you @benmcmorran. This is what I need, it really helps. VSIX working correctly without problem.

BelKed, 2022-03-24:
Thanks, now it works as intended :)

closed 2022-05-03 by github-actions[bot]