microsoft/vscode-arduino #1293: Serial Monitor not working, VSCode 1.57.1, Arduino 0.4.3

scottwoyak, 2021-06-22:
Hi,

I can't get the serial monitor to work  in VSCode. I'm running all the latest tools as of 6/2021. Things work fine in the Arduino IDE. I can compile and upload in VSCode, but nothing displays in the serial monitor output when selected. After compiling and uploading in VSCode, I can open the Arduino IDE and it will display content in its serial monitor.

Version: 1.57.1 (system setup)
Commit: 507ce72
Date: 2021-06-17T13:28:07.755Z
Electron: 12.0.7
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Windows_NT x64 10.0.19042

Arduino Board: Adafruit Feather M0
Baud: 9600 (set in VSCode and my sketch). I've tried 115200 too
COM5 port

Reproduction Steps:
1. Compile and upload any sketch that writes using Serial.println() or equivalent
2. Select the "Serial Monitor" view in the Output window. Nothing is displayed

Is there a way to debug this further? I don't see any serial related content in the Arduino output window with the debug level set to verbose. Maybe a log file somewhere?

ericjhkim, 2021-07-28:
I'm having the same issue as you. I do get "Opening the serial port - COM#" and "Opened the serial port - COM#" but nothing after, even though I have print statements in the sketch.
Perhaps VSCode isn't receiving the data from the arduino?

hoekma, 2021-08-05:
I'm experiencing the same thing with VSCode 1.58.2 and vscode-arduino 0.4.3.    This was working very recently (like maybe a week or two ago) but all of a sudden it's not working today.    I can upload to the device via vscode, but no serial monitor output.  I can see serial output via the Arduino app, so I know the device is generating output.

AlessandroPerazzetta, 2021-08-31:
Same behaviour on VSCodium with vscode-arduino 0.4.4 installed from github (latest release) and also with extensions gallery:

Version: 1.59.1
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-20T00:16:53.906Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Linux x64 5.4.0-81-generic

Boards used: Heltec WiFi Lora 32(V2), Wemos D1 Mini ESP32
Baud: 115200 (/dev/ttyUSB0, /dev/ttyUSB1)

In arduino IDE all works.

adiazulay, 2021-08-31:
@AlessandroPerazzetta can you open that as a new issue? I think there's likely a different cause here.

closed 2021-08-31 by adiazulay

hoekma, 2021-08-31:
@adiazulay Just curious, since you closed it, was the OP's issue (that I also have) resolved for VSCode?

adiazulay, 2021-08-31:
@hoekma v0.4.4 resolves OP's issue. We transitioned to using an external serial monitor, so there is no Node/Electron dependency. If you having an issue getting the serial monitor working in v0.4.4 you can open a new issue and we can figure out why it's not working.

hoekma, 2021-09-02:
@adiazulay Thanks for the fast response.  Tested and still no luck.  Created issue #1338 per your suggestion.