microsoft/vscode-arduino #1298: Unable to select programmer

Juppi88, 2021-07-03:
I have a freshly installed Visual Studio Code with the vscode-arduino extension with a freshly installed Arduino IDE linked to it. I am trying to upload a pretty basic sketch to one of my SparkFun Pro Micro boards.

I am able to select the board I am going to use, as well as the COM port used for programming, However, I am not able to select any Programmer, because for some reason the Visual Studio Code extension isn't able to detect any programmers supported by Arduino:

[image]
[image]

If there are any logs or other information I can provide, please let me know.

ChristianDavis, 2021-07-20:
I had this same issue, but the docs define that as "external" programmers. If I just use the "upload" command it connects to the board via the CLI

afreisinger, 2021-08-16:
I had this same issue.

afreisinger, 2021-08-16:
Solved !
https://randomnerdtutorials.com/solved-arduino-ide-esp32-esp8266-installation/

magnets99, 2021-09-18:
@afreisinger i'm not sure it's the same issue. OP was not having problems with esp. I have the same issue, on linux, so i'm also unable to follow these solution instructions

magnets99, 2021-11-04:
I'm also having the same issue as @mumblety along with selecting the board. it gives me the box, but it is unpopoulated and the the dropdown is disabled (becuase it is empty).

I have tried using the arduino-cli but for some reason code says that my path to the cli is wrong (it is not, i can literally paste the path from the json file into bash and it runs).

magnets99, 2021-11-07:
the issue has been fixed for me now with version 0.4.7
i did have to choose the board first, then the programmer was populated, which i guess makes sense.

benmcmorran, 2022-01-18:
I'm glad this is working for you now! If anyone else in this thread is still having issues, please let me know.

closed 2022-02-24 by github-actions[bot]