microsoft/vscode-arduino #1299: identifier "Serial1" is undefined C/C++(20)

wbadry, 2021-07-04:
**OS**: Windows 10 Version 21H1 (OS Build: 19.043.1081) / UBUNTU 20.04 LTS
**VS Code**:  1.57.1
**Arduino IDE**:  1.8.15
**Arduino Extension**:  0.4.3

# Problem:

The `Serial1` is undefined. The code compiles with no issues though.

[image]

### arduino.json

```json
{
    "sketch": "emulate_data_transfer.ino",
    "board": "arduino:avr:leonardo",
    "output": "../build_emulator_data_transfer"
}
```

### c_cpp_properties.json

```json
{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\bin\\avr-g++",
            "compilerArgs": [
                "-w",
                "-std=gnu++11",
                "-fpermissive",
                "-fno-exceptions",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-Wno-error=narrowing"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\leonardo",
                "C:\\Users\\WALEED ELBADRY\\Documents\\Arduino\\libraries\\TimerOne",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\7.3.0\\include",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\7.3.0\\include-fixed",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\avr\\include"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
                "F_CPU=16000000L",
                "ARDUINO=10815",
                "ARDUINO_AVR_LEONARDO",
                "ARDUINO_ARCH_AVR",
                "USB_VID=0x2341",
                "USB_PID=0x8036",
                "USB_MANUFACTURER=\"Unknown\"",
                "USB_PRODUCT=\"Arduino Leonardo\"",
                "__DBL_MIN_EXP__=(-125)",
                "__HQ_FBIT__=15",
                "__cpp_attributes=200809",
                "__UINT_LEAST16_MAX__=0xffffU",
                "__ATOMIC_ACQUIRE=2",
                "__SFRACT_IBIT__=0",
                "__FLT_MIN__=1.17549435e-38F",
[... 474 lines cut ...]
                "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
                "__LDBL_DIG__=6",
                "__FLT_DECIMAL_DIG__=9",
                "__UINT_FAST16_MAX__=0xffffU",
                "__GCC_ATOMIC_SHORT_LOCK_FREE=1",
                "__BUILTIN_AVR_ABSHK=1",
                "__BUILTIN_AVR_FLASH_SEGMENT=1",
                "__INT_LEAST64_WIDTH__=64",
                "__ULLFRACT_MAX__=0XFFFFFFFFFFFFFFFFP-64ULLR",
                "__UINT_FAST8_TYPE__=unsigned char",
                "__USFRACT_EPSILON__=0x1P-8UHR",
                "__ULACCUM_FBIT__=32",
                "__QQ_IBIT__=0",
                "__cpp_init_captures=201304",
                "__ATOMIC_ACQ_REL=4",
                "__ATOMIC_RELEASE=3",
                "__BUILTIN_AVR_FMUL=1",
                "USBCON"
            ]
        }
    ]
}
```

I tried the solution in #866 and unfortunately, the verification overrides the setting and `Serial1` remains undefined.

DinosDev, 2021-07-13:
Hi Guy! In your code, you writed wrong.
Remove This:
`Serial1.begin(115200)`
And try This
`Serial.begin(115200)`

wbadry, 2021-07-13:
I actually did not !

Arduino Leonardo has 'Serial1' . 'Serial' is the cable port. I am using the TX/RX pins for streaming through another device. 

Thank you for your answer.

2bam, 2021-08-08:
@wbadry I have the same issue with the Mega board (need to access Serial Serial1 and Serial2)
There is a workaround [in this issue](https://github.com/microsoft/vscode-arduino/issues/866#issuecomment-555832272), but you have to patch each time the intellisense file regenerates.

a7hybnj2, 2021-11-28:
I am having this same issue with "MightyCore" and Serial1. I don't much like the quick-fix provided above. Hoping adding traffic to this issue will lead to a better solution.

viaSeunghyun, 2021-12-29:
I have same problem.

My Env is:
VSCode:  1.59.0
Arduino:  1.8.15
Arduino extention: 0.4.8
Target Board: Mega2560
OS: Windows 10.0.18363.1977

h3xcat, 2022-01-20:
This was driving me insane, but I found a solution that would survive intellisense config rebuild by adding the following option inside your `arduino.json`.
```json
"buildPreferences": [["compiler.cpp.extra_flags", "-D__AVR_ATmega2560__"]]
```

You can find macros for your board here.
https://onlinedocs.microchip.com/pr/GUID-317042D4-BCCE-4065-BB05-AC4312DBC2C4-en-US-2/index.html?GUID-830E14F8-A333-4B38-A206-877D7694F324

JakePorter05, 2022-02-07:
[Error.md](https://github.com/microsoft/vscode-arduino/files/8018766/Error.md)

JakePorter05, 2022-02-07:
[Error.md](https://github.com/microsoft/vscode-arduino/files/8018776/Error.md)

JakePorter05, 2022-02-07:
# Another log

I have the same problem with these specs

## Specs

VS Code - 1.64.0  
VS Arduino - 0.4.10

```
{
    "configuration": "cpu=atmega2560",
    "board": "arduino:avr:mega",
    "sketch": "OrderMeg.ino",
    "output": "../../../ArduinoOutput"
}
```

## Code

```
  // Setup Serial
  Serial.begin(115200);
  Serial2.begin(115200);
```

## Error

identifier "Serial2" is undefined

h3xcat, 2022-02-09:
@JakePorter05 Did you try using my solution?

JakePorter05, 2022-02-09:
@h3xcat  Yes your fix worked, but since this is a very mainstream board it would be nice to have it fixed as it still is a bug that shouldn't need a manual modification by the developer.

dennowiggle, 2022-03-11:
@h3xcat Thank you. 

Adding the "buildPreferences" line to arduino.json fixed my many reported errors in intellisense as well.

anzbert, 2023-03-09:
@h3xcat Thx so much. That buildPreferences trick worked beautifully!!

anzbert, 2024-04-23:
i am using the @h3xcat trick, which used to work fine.
now i am getting this error message when compiling:

```In file included from /Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBAPI.h:44:0,
                 from /Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.cpp:20:
/Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.cpp:75:29: error: 'USB_VID' was not declared in this scope
  D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
                             ^
/Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.h:270:61: note: in definition of macro 'D_DEVICE'
  { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                             ^~~~
/Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.cpp:75:29: note: suggested alternative: 'USB_Recv'
  D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
                             ^
/Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.h:270:61: note: in definition of macro 'D_DEVICE'
  { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                             ^~~~
/Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.cpp:75:37: error: 'USB_PID' was not declared in this scope
  D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
                                     ^
/Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.h:270:66: note: in definition of macro 'D_DEVICE'
  { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                  ^~~~
/Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.cpp:75:37: note: suggested alternative: 'USB_Recv'
  D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
                                     ^
/Users/anz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/USBCore.h:270:66: note: in definition of macro 'D_DEVICE'
  { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                  ^~~~

Error during build: exit status 1```

anzbert, 2024-04-23:
i've eliminated the error now like this:

```
#ifndef __INTELLISENSE__
  Serial1.begin(31250); // Set MIDI baud rate
#endif
```