microsoft/vscode-arduino #1301: Error installing dummy

jerryc05, 2021-07-14:
A fresh install of `vscode-arduino v0.4.3` and a fresh install of `arduino-cli`  of `nightly-20210714 Commit: e7d4eaab Date: 2021-07-14T01:21:31Z`

When I clicked `Refresh Library Index`, I got these error in the `OUTPUT` panel:
```
[Starting] Update package index files...
dummy

 0 / ?    0.00%
Updating index: library_index.json.gz downloaded

 0 / 542    0.00%
Updating index: library_index.json.sig downloaded

 0 / 399148    0.00%
Updating index: package_index.json downloaded

 0 / 543    0.00%
Updating index: package_index.json.sig downloaded
Downloading missing tool builtin:ctags@5.8-arduino11...

 0 / 116455    0.00%
builtin:ctags@5.8-arduino11 downloaded
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.3.0-rc1...

 0 / 1709333    0.00%
builtin:serial-discovery@1.3.0-rc1 downloaded
Installing builtin:serial-discovery@1.3.0-rc1...
builtin:serial-discovery@1.3.0-rc1 installed
Invalid argument passed: invalid item dummy
[Error] Exit with code=7

[Starting] Update library index files...
Error installing dummy: Error resolving dependencies for dummy@: looking for library: library dummy not found
[Done] Updated library index files.
```

odahcam, 2022-07-30:
Any workarounds on that?

strazto, 2022-10-22:
Related https://github.com/microsoft/vscode-arduino/issues/1388 

@odahcam yep, heres one https://github.com/microsoft/vscode-arduino/issues/1388#issuecomment-1010547815 

> I had the same error output:
> 
> ```
> [Starting] Update package index files...
> dummy
> Invalid argument passed: invalid item dummy
> [Error] Exit with code=7
> ```
> 
> This is what I did to workaround the issue:
> 
>     1. Added the additional urls directly to `arduino-cli` config by executing:
> 
> 
> ```
> arduino-cli config init --additional-urls https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
> ```
> 
>     2. Executed `arduino-cli core update-index`. You'll see the following output:
> 
> 
> ```
> Updating index: package_index.json downloaded                                                                                                                   
> Updating index: package_index.json.sig downloaded                                                                                                               
> Updating index: package_esp32_index.json downloaded
> ```
> 
> After performing the above, I could view the `esp32` package in in the Arduino Board Manager and install it successfully.
> 
> My `settings.json`
> 
> ```
> {
>     "arduino.path": "/opt/homebrew/bin/",
>     "arduino.useArduinoCli": true,
>     "arduino.commandPath": "arduino-cli",
>     "arduino.additionalUrls": "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json"
> }
> ```
> 
> Mac OS `11.6.2` (Big Sur) on M1 VS Code `1.63.2` VS Code Arduino Extension `0.4.8 preview` Arduino CLI `0.20.2` installed using Homebrew
> 
> Hope this helps.