microsoft/vscode-arduino #1308: Latest VS Code regular release breaks the serial monitor

mathieupothier, 2021-08-05:
Hello,

The latest version of VSCode (released today) seems to break the serial monitor with the following error.

VS code data:
Version: 1.59.0 (system setup)
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:13:12.822Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Windows_NT x64 10.0.19043

Extension v0.4.3

Error message when trying to open the serial monitor:

The module '\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node'

was compiled against a different Node.js version using
NODE_MODULE_VERSION 87. This version of Node.js requires
NODE_MODULE_VERSION 89. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Leo-Z0, 2021-08-06:
Same issue here (running Windows 10)

Leo-Z0, 2021-08-06:
> Same issue here (running Windows 10)

Rolling back to VS Code [version 1.58](https://code.visualstudio.com/updates/v1_58) helps

Llaves, 2021-08-07:
at the risk of the blindingly obvious, don't forget you can use the Arduino IDE Serial monitor until this is fixed if you don't want to roll back VS Code.

rrakso, 2021-08-09:
I've also faced this issue :/ - as always...

adiazulay, 2021-08-16:
tracking this with #1312

closed 2021-08-16 by adiazulay