microsoft/vscode-arduino #1310: The module usb-detection in VSCode Arduino 0.4.3 was compiled different Node.js version

viaSeunghyun, 2021-08-06:
I installed ㅍ0.4.3

This extension worked fine for the first 2 days.
I assigned an additional shortcut to this extension yesterday, and after doing some additional development work, I shut down the PC.
Today I faced the below error message, the extension is no longer available.
Even after reinstalling and restarting!
Currently, only the IntelliSense Rebuild and Compile functions work for the extension function of my pc.
This message either shows up on all problematic menu, or doesn't respond at all.
I did not install a separate Node.js version, and there is no separate Node.js anywhere in the path.

```
Failed to open serial port COM5 due to error: + Error: The module '\\?\c:\Users\bluems\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 87. This version of Node.js requires
NODE_MODULE_VERSION 89. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

simon88, 2021-08-06:
Same error for me

viaSeunghyun, 2021-08-06:
> Same error for me, after vscode update. Compile and upload works fine.
> 
> `Failed to open serial port /dev/tty.usbmodem14201 due to error: + Error: The module '/Users/simon/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/usb-detection/build/Release/detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 87. This version of Node.js requires NODE_MODULE_VERSION 89. Please try re-compiling or re-installing the module (for instance, using `npm rebuild`or`npm install`).`

adiazulay, 2021-08-16:
tracking this with #1312

closed 2021-08-16 by adiazulay