microsoft/vscode-arduino #1312: Serial monitor not works after VSCode updated to 1.59.0

davcolme, 2021-08-06:
Today an update has been automatically installed, I think the version 1.59.0 and Serial monitor not works after VSCode updated. Error:
Failed to open serial port COM7 due to error: + Error: The module '\\?\c:\Users\****\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 87. This version of Node.js requires
NODE_MODULE_VERSION 89. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

julesgilson, 2021-08-07:
@davcolme  V1.59.0 is the version of VSC, the current version of the Arduino extension is 0.4.3 and that is the one you have (you can see in the error output).

I have a similar issue - haven't used this extension for some months, so can't say if it is due to a recent update, but for sure I cannot select COM port from the status bar (no selection popup appears) and if I set it manually in arduino.json I get this error when trying to open it in the monitor

``` Failed to open serial port COM4 due to error: + Error: \\?\c:\Users\xyz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application. \\?\c:\Users\xyz\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node ```

So some dependency issues

- Windows 10 64bit (v2004)
- VSC v1.59.0
- Arduino Ext. v0.4.3
- Node v14.17.3

julesgilson, 2021-08-07:
Seeing as it is a Saturday... the dependency node-usb-native has its own dependency usb-detection, which is the one that is leading to all the issues (I just read all the other recent issues raised) - that repo is here:

https://github.com/MadLittleMods/node-usb-detection

It highlights its need for prebuilt binaries leading to node clashes with Electron and points to this:

https://www.electronjs.org/docs/tutorial/using-native-node-modules

davcolme, 2021-08-07:
> @davcolme V1.59.0 is the version of VSC, the current version of the Arduino extension is 0.4.3 and that is the one you have (you can see in the error output).
>
> I have a similar issue - haven't used [...]

The problem is surely due to an update. I was using the serial monitor, it asked me to restart to complete an update and the serial port stopped working. I do not know what update was installed.

Llaves, 2021-08-07:
@julesgilson - 
Following your links, I've successfully installed electron-rebuild. However, when I run electron-rebuild.cmd (I'm on windows), I get an error - 
```
An unhandled error occurred inside electron-rebuild
Unable to find electron's version number, either install it or specify an explicit version
```
I'm guessing this is occurring because I need to run the cmd in a specific directly relative to VS Code or the extension, but I have no clue what that might be. 

Any guidance is welcome.

julesgilson, 2021-08-09:
> 
> Any guidance is welcome.

Sorry, I wasn't very clear - I just posted that info for the extension devs, to hopefully give them a head start in fixing the issue.

rrakso, 2021-08-09:
#1308

imlinhanchao, 2021-08-10:
I think this issue was the same issue in my extension. I have rebuilded the node-usb-native for electron 13.1.7. It's worked now. Maybe you can try download the node module from (here)[https://github.com/imlinhanchao/vsc-serialport-helper/tree/master/src/script/node] and put it in /your_user_path/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/node-usb-native/lib/native.

julesgilson, 2021-08-10:
I believe all the devs need to do is rebuild the extension after rebuilding the binaries for Electron v13 that are dependencies of the USB detector package.

I am not sure if in-place replacement of the dependencies will work without a rebuild of the extension itself... worth a try I guess

TristeFigure, 2021-08-11:
> electron-rebuild

This is the way.

- Use nodejs < 16 (causes problem with node-gyp and the v8 c typedefs). For instance
  - `nvm install v15.14.0`
  - `nvm use v15.14.0`
- Then install `electron-rebuild`
  - `npm install electron -g` (you may need to run `npm update node-abi` additionally)
  - `npm install electron-rebuild -g`
- Rebuild `usb-detection`
  - `cd ~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/usb-detection/`
  - `electron-rebuild`
- Rebuild `@serialport/bindings` (yep, there is another package to fix 🙀)
  - cd `~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/@serialport/bindings`
  - `electron-rebuild`
  
  Everything's back on tracks after those moves.

Llaves, 2021-08-12:
@TristeFigure  - 
I followed your instructions (adapted for a Windows environment.)
I had already downloaded the latest windows installer, which is 14.17.4. 

When I run 
```
electron-rebuild.cmd
```
in the usb-detection directory I get an error about version, so I add a -v flag to get past that.
````
electron-rebuild.cmd -v 13.1.9
```

But then I get
```
 fatal error C1083: Cannot open include file: 'nan.h': No such file or directory
```

My node.js version is older than your instructions - maybe that's my problem?

julesgilson, 2021-08-12:
> electron-rebuild.cmd

On Windows at least, you will have needed to install all the build tools first, Python, VS etc.

Hopefully the devs will rebuild their extension and release a new version...

julesgilson, 2021-08-12:
@Llaves 

As you most likely have the same environment as me, you can download the rebuilt dependencies from the link below.

http://iethis.co.uk/rebuilt-usb-dependencies.zip

Extract them and copy them over the existing folders in

` C:\Users\<user>\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules `

By all means make a copy of the originals first.

Note: This will likely only work with the following env:

- Windows 10 64
- NodeJS v14.17.x
- VSC v1.59.0 (more importantly, the Electron version it uses, v13.1.7)

Llaves, 2021-08-12:
@julesgilson 
Thanks for taking the time to answer my question and send the rebuilt dependencies. Unfortunately, this did not solve my problem. I'm still getting the version 87 vs 89 error msg. 
Per your env requirements - I'm running Win 10 64, NodeJS 14.17.4, VSC 1.59.0/Electron 13.1.7. However Help->About shows node 14.16.0
but if I open the terminal window in VS Code and type
```
node -v
```
I get 14.17.4

How can I force VS Code to use the newer NodeJS?

Of course, I can just wait for the extension to be updated.

TristeFigure, 2021-08-12:
I think `NODE_MODULE_VERSION = 89` only comes with electron as described in the table at the bottom of this [link](https://www.npmjs.com/package/electron-releases) all the more so it's missing from [this table](https://nodejs.org/en/download/releases/).

>  fatal error C1083: Cannot open include file: 'nan.h': No such file or directory

I had the same error and fixed it by reinstalling node-gyp. As a Mac user I had problems with the tool chain I fixed following this guide https://github.com/nodejs/node-gyp/blob/master/macOS_Catalina.md.

To install node-gyp: `npm install -g node-gyp`.

The error I had with node>=16.0.0 was in V8.h, something about `sometype_cv`  vs. `sometype_cv_t`.

I tried compiling with different versions of node but always had a mismatch with `NODE_MODULE_VERSION` until I realized I had to compile from within electron. Thanks to @ Llaves for bringing up `electron-rebuild`.

imlinhanchao, 2021-08-13:
@Llaves Maybe you can use the node module that I build. You can get it from [here](#issuecomment-895749851) . I was use node-gyp to build. Build command is `node-gyp rebuild --target=13.1.7 --arch=x64 --dist-url=https://atom.io/download/electron`. I have try it worked well.

orlade, 2021-08-14:
In addition to @TristeFigure 's wonderful guidance, I also had to run `yarn install` in `~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/usb-detection/` to resolve the error:
```
⠋ Building module: usb-detection, Completed: 0In file included from ../src/detection.cpp:1:
../src/detection.h:13:10: fatal error: 'nan.h' file not found
#include <nan.h>
         ^~~~~~~
```

Hiro-Studying, 2021-08-14:
https://github.com/microsoft/vscode-arduino/issues/1312#issuecomment-897224945
https://github.com/microsoft/vscode-arduino/issues/1312#issuecomment-898814632
Thanks! Works for me!

Llaves, 2021-08-15:
@orlade - installing and running yarn solved my issue with the nan.h error. Thanks to everyone who has contributed to getting this running again.

adiazulay, 2021-08-16:
Hi all, I just returned from vacation today. I should have a fix for this out in the next day or 2.

adiazulay, 2021-08-20:
Hey folks! We have an release candidate that uses a new serial monitor. This fixes the current issue and will avoid this issue from happening in the future.

You can download the rc [here](https://github.com/microsoft/vscode-arduino/releases/tag/v0.4.4-rc1). And install it manually with `Extension: Install from VISX...`

hdpklm, 2021-08-20:
> Hey folks! We have an release candidate that uses a new serial monitor. This fixes the current issue and will avoid this issue from happening in the future.
>
> You can download the rc [...]

thank you so much 🌹

dtboy1995, 2021-08-21:
> Hey folks! We have an release candidate that uses a new serial monitor. This fixes the current issue and will avoid this issue from happening in the future.
>
> You can download the rc [...]

deply output not work... All outputs are non-delay

basbloemsaat, 2021-08-22:
It works for me, selecting com port works and serial works on ESP32-dev and ESP8266 (D1 mini). Thanks!

frascuelillo, 2021-08-23:
> Hey folks! We have an release candidate that uses a new serial monitor. This fixes the current issue and will avoid this issue from happening in the future.
>
> You can download the rc [...]

Also works for me, Thank's a lot!!

RMLEARNEY, 2021-08-23:
The rc presents this new error message:
`Unhandled exception when cleaning up build "Uploading": {}`

hdpklm, 2021-08-23:
> The rc presents this new error message:
> `Unhandled exception when cleaning up build "Uploading": {}`

same error

adiazulay, 2021-08-23:
Do you have any repro steps for the error? It showed up once for me, but I can't get it show again.

hdpklm, 2021-08-23:
> Do you have any repro steps for the error? It showed up once for me, but I can't get it show again.

when i click on upload.
i have another issue when i click on select programer for esp32, show an empty list, but for arduino uno show a lot of options

anoff, 2021-08-24:
Not sure if this is a bug - but I remember the behavior being different before. Running the `0.4.4-rc1`

For me the Serial Monitor does not pick up on data if I hard reset the board. What works:
1. Open Serial Monitor
2. Flash
3. See "new" output after flashing restarts the device

What also works:
1. Restart via HW button
2. Open Serial Monitor quickly
3. see output

What does not work:
1. Serial Monitor open
2. see output
3. Restart via HW button
4. no more output

If I disable serial monitor in VSCode and use Arduino IDE serial monitor everything works as expected (serial traffic is picked up even after hard restart)

adiazulay, 2021-08-24:
Hi Folks,

We have a new RC today: https://github.com/microsoft/vscode-arduino/releases/tag/v0.4.4-rc2

If there are no major issues I'll release it end of day tomorrow to the marketplace.
@anoff Can you open that as a new issue. I don't want that hold up the release, but I also don't want it to get lost.

YuMERA, 2021-08-24:
> Hi Folks,
>
> We have a new RC today: https://github.com/microsoft/vscode-arduino/releases/tag/v0.4.4-rc2
>
> If there are no major issues I'll release it end of day tomorrow to the marketplace.
> @anoff [...]

Failed to open serial port COM10 due to error: + Error: spawn UNKNOWN

adiazulay, 2021-08-24:
@YuMERA can provide more details? What OS are you on, what version of VSCode are you using?

YuMERA, 2021-08-24:
Version: 1.59.1 (user setup)
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-19T11:56:07.791Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Windows_NT ia32 10.0.19043

solariz, 2021-08-25:
Unfortunately similar here.

If I want to select a COM by clicking on the com port footer nothing happens, no error, no popup, nothing. If I manually add the com port to the arduino.json it still throw me the not native win32 application error:

`Failed to open serial port COM7 due to error: + Error: \\?\c:\Users\mg\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application. \\?\c:\Users\mg\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node`

```
Version: 1.59.1 (user setup)
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-19T11:56:46.957Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Windows_NT x64 10.0.19043
```