microsoft/vscode-arduino #1315: Serial port

CodeNameHawk, 2021-08-10:
Issue Type: <b>Bug</b>

can`t select serial port.

Extension version: 0.4.3
VS Code version: Code 1.59.0 (379476f0e13988d90fab105c5c19e7abc8b1dea8, 2021-08-04T23:13:20.182Z)
OS version: Linux x64 5.4.0-80-generic
Restricted Mode: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Pentium(R) Dual-Core CPU       T4500  @ 2.30GHz (2 x 2294)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: disabled_off<br>oop_rasterization: disabled_off<br>opengl: enabled_on<br>rasterization: disabled_software<br>skia_renderer: enabled_on<br>video_decode: disabled_software<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|1, 2, 1|
|Memory (System)|3.81GB (0.57GB free)|
|Process Argv|--no-sandbox --unity-launch --crash-reporter-id 02ee2987-17b5-4cfb-9854-12b20aba17e2|
|Screen Reader|no|
|VM|0%|
|DESKTOP_SESSION|xubuntu|
|XDG_CURRENT_DESKTOP|XFCE|
|XDG_SESSION_DESKTOP|xubuntu|
|XDG_SESSION_TYPE|x11|
</details>

wbadry, 2021-08-10:
Hello @CodeNameHawk 

A quick check, 

Have you enabled Arduino permission to access the serial port?

```bash
sudo usermod -a -G dialout $USER
```
[image]

CodeNameHawk, 2021-08-11:
Yes.

Serial work in ArduinoIDE.

If add to arduino.json
"port": "/dev/ttyUSB0"
and try open serial monitor see:

Failed to open serial port /dev/ttyUSB0 due to error: + Error: The module
'/home/marian/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/usb-detection/build/Release/detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 87. This version of Node.js requires
NODE_MODULE_VERSION 89. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

ETLin, 2021-08-11:
same prob. 

  System Version:	macOS 11.5 (20G71)
  Kernel Version:	Darwin 20.6.0

Failed to open serial port /dev/tty.usbserial-14340 due to error: + Error: dlopen(/Users/MyName/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/usb-detection/build/Release/detection.node, 1): no suitable image found. Did find: /Users/MyName/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/usb-detection/build/Release/detection.node: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00 /Users/MyName/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.3/out/node_modules/usb-detection/build/Release/detection.node: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00

ETLin, 2021-08-11:
The Visual Code is compiled by Node.js: 14.16.0.
But the build version of usb-detection needs NODE_MODULE_VERSION 89.

Version: 1.59.0 (Universal)
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:14:40.191Z (6 days ago)
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Darwin x64 20.6.0

ETLin, 2021-08-11:
Node.js: 14.16.0. => NODE_MODULE_VERSION=83

https://nodejs.org/en/download/releases/

DzikuVx, 2021-08-13:
Same here

```
Failed to open serial port COM30 due to error: + Error: The module '\\?\c:\Users\pspyc\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 87. This version of Node.js requires NODE_MODULE_VERSION 89. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).
```

fscheu, 2021-08-13:
Same issue here. Is there a viable workaround?

DzikuVx, 2021-08-13:
Downgrade to vscode 1.58.2

solariz, 2021-08-16:
Same Error here.

`Failed to open serial port COM3 due to error: + Error: \\?\c:\Users\mg\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application. \\?\c:\Users\mg\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node`

vscode 1.59.0
node 14.16
Win10 x64

Unfortunately cant downgrade node due to other incompatibilities. Will go back to Arduiono IDE so long.

adiazulay, 2021-08-16:
Tracking this with issue #1312

closed 2021-08-16 by adiazulay

hyshiah, 2021-11-24:
For raspberryPi 4 using raspberry os... the probram is 
/home/pi/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.8/out/serial-monitor-cli/linux/main is build for x86
arm user need rebuild by yourself.
download https://github.com/microsoft/serial-monitor-cli.git
and follow readme.md to build "main" for your raspberryPi
and copy "main" to 
/home/pi/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.8/out/serial-monitor-cli/linux

for me, every thing ok.