microsoft/vscode-arduino #1316: [REQUEST] Configure build folder for Arduino

wbadry, 2021-08-10:
Hello,
It would be nice if there is a way to configure the path for the `build` folder, `debug` or `release` to make it easier to get the `hex` or `elf` file in a directory within the project folder like C++

```
motor_speed
│   monitor_speed.ino
│
├───.vscode
│       arduino.json
│       c_cpp_properties.json
│
└───build
    └───release
        │   build.options.json
        │   includes.cache
        │   monitor_speed.ino.eep
        │   monitor_speed.ino.elf
        │   monitor_speed.ino.hex
        │   monitor_speed.ino.with_bootloader.bin
        │   monitor_speed.ino.with_bootloader.hex
        │
        ├───core
        │       abi.cpp.d
        │       abi.cpp.o
        │       CDC.cpp.d
        │       CDC.cpp.o
        │       core.a
        │       HardwareSerial.cpp.d
        │       HardwareSerial.cpp.o
        │       HardwareSerial0.cpp.d
        │       HardwareSerial0.cpp.o
        │       HardwareSerial1.cpp.d
        │       HardwareSerial1.cpp.o
        │       HardwareSerial2.cpp.d
        │       HardwareSerial2.cpp.o
        │       HardwareSerial3.cpp.d
        │       HardwareSerial3.cpp.o
        │       hooks.c.d
        │       hooks.c.o
        │       IPAddress.cpp.d
        │       IPAddress.cpp.o
        │       main.cpp.d
        │       main.cpp.o
        │       new.cpp.d
        │       new.cpp.o
        │       PluggableUSB.cpp.d
        │       PluggableUSB.cpp.o
        │       Print.cpp.d
        │       Print.cpp.o
        │       Stream.cpp.d
        │       Stream.cpp.o
        │       Tone.cpp.d
        │       Tone.cpp.o
        │       USBCore.cpp.d
        │       USBCore.cpp.o
        │       WInterrupts.c.d
        │       WInterrupts.c.o
        │       wiring.c.d
        │       wiring.c.o
        │       wiring_analog.c.d
        │       wiring_analog.c.o
        │       wiring_digital.c.d
        │       wiring_digital.c.o
        │       wiring_pulse.c.d
        │       wiring_pulse.c.o
        │       wiring_pulse.S.d
        │       wiring_pulse.S.o
        │       wiring_shift.c.d
        │       wiring_shift.c.o
        │       WMath.cpp.d
        │       WMath.cpp.o
        │       WString.cpp.d
        │       WString.cpp.o
        │
        ├───libraries
        │   ├───Encoder
        │   │       Encoder.cpp.d
        │   │       Encoder.cpp.o
        │   │
        │   └───TimerOne
        │           TimerOne.cpp.d
        │           TimerOne.cpp.o
        │
        ├───preproc
        │       ctags_target_for_gcc_minus_e.cpp
        │
        └───sketch
                monitor_speed.ino.cpp
                monitor_speed.ino.cpp.d
                monitor_speed.ino.cpp.o
```

[image]

m-mcgowan, 2021-08-24:
I agree. I have spent a few minutes searching and don't see where the files get written to, so it's not completely obvious.

m-mcgowan, 2021-08-24:
For anyone else wondering, you can add `"output" : "build"` to the `arduino.json` file. 

https://github.com/microsoft/vscode-arduino/issues/1302

wbadry, 2021-08-24:
[image]

[image]

closed 2021-08-24 by wbadry