microsoft/vscode-arduino #1319: Upload command sometimes fails because of an unknown flag

soul-codes, 2021-08-15:
Platform: Ubuntu 20.10.
Arduino version: CLI 0.13.0.

Using the `Arduino CLI: Upload` command sometimes results in the following error in the console:

```
Error: unknown flag: --build-path
Usage:
  arduino-cli upload [flags]
```

This seems to happen after the first succesful upload. I have `output` configured in `arduino.json`.

piif, 2022-01-16:
I just got same issue, with arduino-cli v0.20.2 and arduino extension v0.4.10

"CLI:Upload" command works after replacing `--build-path` by `--input-dir` at line 1012 of `.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/src/arduino/arduino.js` but then the "Upload" button fails.
It's because first one calls `arduino-cli upload` command which accept `input-dir` option, but second uses `arduino-cli compile --upload` which accepts `build-path` options ...

gasparo, 2022-04-27:
This worked for me too.. Please change that on next release.

cdealti, 2022-11-14:
Still present with Visual Studio Code extension for Arduino v0.4.12 with:
* Arduino IDE v2.0.1
* arduino-cli 0.28.0

craigdetter, 2023-01-16:
This exact error is occurring on these versions:

Arduino IDE v2.0.3
arduino-cli  Version: 0.29.0 Commit: 76251df9 Date: 2022-11-16T14:49:42Z
VSCode: Version: 1.74.3
Commit: 97dec172d3256f8ca4bfb2143f3f76b503ca0534
Date: 2023-01-09T17:05:47.808Z (1 wk ago)

[VSCode Error- Unknown flag.txt](https://github.com/microsoft/vscode-arduino/files/10429953/VSCode.Error-.Unknown.flag.txt)

dancesWithMachines, 2023-01-26:
Issue still present with extension v0.4.13

Addensess, 2023-04-26:
I use the bundled cli and still get this with v0.6.0

GalBrandwine, 2023-06-25:
Same for Version 0.6.0

janoist1, 2023-12-21:
I'm getting the error too - late 2023.

Tynopia, 2023-12-21:
I'm getting this issue too

willlogs, 2024-03-05:
Can't believe this bug is still present! Just change the flag

willlogs, 2024-04-16:
Clean up arduino cli installation and use something other than stupid snapd
to install it. That fixed it for me

wrenchpilot, 2024-04-24:
I just ran into this when trying to use `CTRL-SHIFT-P` -> `Arduino CLI: Upload Using Programmer`

Try using `CTRL-SHIFT-P` -> `Arduino: Upload Using Programmer` (i.e. Don't use the Arduino CLI)

roblatour, 2024-07-06:
I am having this problem too.  
I am building on a windows machine and have a space in my user name - this has cause other problems in the past - could it be causing this too?