microsoft/vscode-arduino #1320: unable to select com port

Mega033, 2021-08-17:
Issue Type: <b>Bug</b>

In visual studio code run select port cmd, my result is whole lot of nothing happens

Extension version: 0.4.3
VS Code version: Code 1.59.0 (379476f0e13988d90fab105c5c19e7abc8b1dea8, 2021-08-04T23:13:12.822Z)
OS version: Windows_NT x64 10.0.18363
Restricted Mode: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-6820HQ CPU @ 2.70GHz (8 x 2712)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.79GB (7.19GB free)|
|Process Argv|--crash-reporter-id 77654c11-6bca-4e0f-a654-3a03462f530a|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383:30185418
pythonvspyt602:30300191
vspor879:30202332
vspor708:30202333
vspor363:30204092
vstes627:30244334
pythonvspyt639:30300192
pythontb:30283811
pythonptprofiler:30281270
vshan820:30294714
vstes263:30335439
pythondataviewer:30285071
pythonvsuse255:30340121
vscod805:30301674
pythonvspyt200:30340761
vscextlang:30333561
binariesv615:30325510
vsccppwt:30329788
pythonvssor306:30344512
bridge0708:30335490
vstre464:30350172
bridge0723:30353136

```

</details>

zcm2005, 2021-08-17:
I met the same problem. Wether I use the command or click the button at the right bottom of VS Code, just nothing happens. But I can successfully upload my sketch using Arduino IDE.

Fergo, 2021-08-18:
Also having the same issue.

Extension version: 0.4.3

VSC Info:
Version: 1.59.0 (user setup)
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:13:12.822Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Windows_NT x64 10.0.19043

CSTHenry, 2021-08-18:
Also having the same issue.

FuentesFj, 2021-08-18:
Here the same issue.
Version: 1.59.0 (user setup)
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:13:12.822Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Windows_NT x64 10.0.19042

It happened to me just after upgrading VisualStudioCode, with the previous version It was working correctly.

alexokaban, 2021-08-19:
Downgrading to previous version ([1.58 from June ](https://code.visualstudio.com/updates/v1_58)) solves the issue.

hdpklm, 2021-08-20:
> Downgrading to previous version ([1.58 from June ](https://code.visualstudio.com/updates/v1_58)) solves the issue.

i try to downgrading vscode and disable auto update. but its update first time is runing
so i cant try the solution.
thank you

alexokaban, 2021-08-20:
You could unplug your Internet first. I downgrading, didn't note the update
and the issue disappeared.

adiazulay, 2021-08-20:
We're tracking this issue with #1312

closed 2021-08-20 by adiazulay