microsoft/vscode-arduino #1324: Can't change serial port line ending sequence

roninbar, 2021-08-21:
In #972 it says:

> Thanks for your contributions. Line endings selection has been removed and replaced by default ending with the carriage return and the line feed `\r\n`.

_Originally posted by @hellyzh in https://github.com/microsoft/vscode-arduino/issues/972#issuecomment-604878823_

but it doesn't say how to change this setting to any other value, e.g. `none`. I went over all the settings and all the commands and couldn't find a way to do it.

JaredTheWolf, 2022-01-01:
This needs some attention. Without being able to change line endings, source code needs to be changed to account for newlines, just because this plugin no longer supports changing the line ending. There seems to be no reasoning as to why this was changed, and makes the serial monitor effectively useless in my case.

gcampbell-msft, 2023-02-16:
@roninbar We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue or if the issue still persists!

closed 2023-02-16 by gcampbell-msft

roninbar, 2023-02-17:
@gcampbell-msft Yes, it works! Thanks!