microsoft/vscode-arduino #1325: Serial port fails to open, Serial port cannot be selected.

iqapps, 2021-08-23:
Issue Type: <b>Bug</b>

Last update of VS-Code breaks Arduino extension - again.

When clicking COM port in status line, nothing happen. When trying to open COM port, ig get this error:

Failed to open serial port COM9 due to error: + Error: The module '\\?\c:\Users\hl\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.3\out\node_modules\usb-detection\build\Release\detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 87. This version of Node.js requires NODE_MODULE_VERSION 89. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).

Extension version: 0.4.3
VS Code version: Code 1.59.1 (3866c3553be8b268c8a7f8c0482c0c0177aa8bfa, 2021-08-19T11:56:46.957Z)
OS version: Windows_NT x64 10.0.19042
Restricted Mode: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i9-9900 CPU @ 3.10GHz (16 x 3096)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|31.79GB (18.42GB free)|
|Process Argv|--crash-reporter-id 937b32ac-d42d-4d5c-8600-a5390f81789c|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383cf:30185419
pythonvspyt700cf:30270857
pythonvspyt602:30300191
vspor879:30202332
vspor708:30202333
vspor363:30204092
pythonvspyt639:30300192
pythontb:30283811
pythonptprofiler:30281270
vshan820:30294714
vstes263:30335439
vscorecescf:30322572
pythondataviewer:30285071
pythonvsuse255:30340121
vscod805:30301674
pythonvspyt200:30340761
vscextlang:30333561
binariesv615:30325510
pythonvssor306:30344512
bridge0708:30335490
vstre464:30350172
bridge0723:30353136
vsdyn321cf:30354108

```

</details>

iqapps, 2021-08-23:
The error description suggest to run `npm (install|rebuild)`

Could there be relevant suggestions to Windows users also.

Scannow, 2021-08-23:
Same error message here, I'm on Manjaro 21, Arduino extension version 0.4.3, VSCode 1.59.0 and Arduino 1.8.15.

anoff, 2021-08-24:
See https://github.com/microsoft/vscode-arduino/issues/1312#issuecomment-902908608 for a release candidate of 0.4.4 that fixes it

adiazulay, 2021-08-24:
as noted we're tracking this with #1312

closed 2021-08-24 by adiazulay