microsoft/vscode-arduino #1335: Serial console extremely slow

DzikuVx, 2021-08-29:
After upgrading from `0.4.3` and VSCode `1.58.2` to `0.4.4` and latest VSCode, Serial console is extremely slow.

Yes, it works, there are no errors but it's a few times slower than before. 

Example 1: `0.4.3`

https://user-images.githubusercontent.com/966811/131247387-fe285a5f-35c2-4856-b403-f2ea270cd9e2.mp4

Example 2: `0.4.4`

https://user-images.githubusercontent.com/966811/131247390-c1e9ce7e-ff77-4ee9-9e68-67048877a832.mp4

Not only ESP32 failed to reset on connect, it's approx 10 times slower than before.

adiazulay, 2021-08-30:
Is it only slow on start up or does it stay slow after it's printing? It's hard from me to tell from the videos.

I'm working to add the RTS and DTR signals into the new serial monitor so the boards should be restarting on connect in the next release of the serial monitor.

vlastahajek, 2021-08-31:
Opened Serial Console also causes a high load of CPU (around 30%).

adiazulay, 2021-08-31:
I think I figured out what causes the high CPU load. It should be resolved in the next release of the Serial Monitor Cli. I hope this also fixes the slow performance.

adiazulay, 2021-08-31:
Can you try uninstalling the extension, restarting VS Code, and installing it again?

vlastahajek, 2021-09-01:
I tried that, but it still causes a high load.

Another new issue is that it shows only new full lines, not char by char prints. (I will create a new issue for this).

vlastahajek, 2021-09-22:
I can confirm this is fixed in 0.4.5

snympi, 2021-09-26:
I can also confirm it is fixed in 0.4.5, thanks @adiazulay

adiazulay, 2021-09-27:
Thanks for confirm this!

closed 2021-09-27 by adiazulay