microsoft/vscode-arduino #1337: Serial Monitor not working on Linux + VSCodium 1.59.1 and vscode-arduino 0.4.4

AlessandroPerazzetta, 2021-08-31:
Reproduction Steps:

1. compile and upload sketch to the board (works fine, verified through external serial moitor debugger and from Arduino IDE)
2. attach to serial monitor and nothing is printed out as expected (ex: Serial.println())

Installed vscode-arduino 0.4.4 from github (latest release https://github.com/microsoft/vscode-arduino/releases/download/v0.4.4/vscode-arduino-0.4.4.vsix) and also with extensions gallery:
Also tried to remove and reinstall extension, nothing changed.

Tried also to look at the VSCode logs (Log (Main), Log (Extension Host)) nothing relevant appear.

Version: 1.59.1
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-20T00:16:53.906Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Linux x64 5.4.0-81-generic

Boards used: Heltec WiFi Lora 32(V2), Wemos D1 Mini ESP32
Baud: 115200 (/dev/ttyUSB0, /dev/ttyUSB1)

In arduino IDE all works.

Eggn1n3, 2021-09-01:
Same here. /dev/ttyUSB0 does sometimes works (more often not). Using an Arduino Node MCU1.0
No error, just an empty serial port while Arduino IDE or other terminal program does show output.

Version: 1.59.1
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-19T11:54:37.513Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Linux x64 5.4.0-81-generic

AlessandroPerazzetta, 2021-09-07:
Just a little update after looking to arduino.log, seems that there's an error related to missing library while trying to open serial monitor:

{"correlationId":"36919a7eda83489999e32dd6638b8a88","command":"arduino.selectSerialPort","message":"executeCommandError","errorLine":"serial-monitor-cli/linux/libpython3.8.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/alessandro/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.4.4","telemetry":true,"level":"error","timestamp":"2021-09-07T18:42:25.572Z"}
{"correlationId":"43929728e80b4a48ac9840e8c94e82c5","command":"arduino.selectSerialPort","message":"executeCommandError","errorLine":"serial-monitor-cli/linux/libpython3.8.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/alessandro/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.4.4","telemetry":true,"level":"error","timestamp":"2021-09-07T18:42:26.952Z"}

System info:

root@KG:/home/alessandro$ cat /etc/issue
Linux Mint 19.3 Tricia \n \l

root@KG:/home/alessandro$ uname -a
Linux KG 5.4.0-81-generic #91~18.04.1-Ubuntu SMP Fri Jul 23 13:36:29 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Library info:

alessandro@KG:~$ dpkg -l |grep libc
ii  klibc-utils                                2.0.4-9ubuntu2                                                                  amd64        small utilities built with klibc for early boot
ii  libc-bin                                   2.27-3ubuntu1.4                                                                 amd64        GNU C Library: Binaries
ii  libc-dev-bin                               2.27-3ubuntu1.4                                                                 amd64        GNU C Library: Development binaries
ii  libc6:amd64                                2.27-3ubuntu1.4                                                                 amd64        GNU C Library: Shared libraries
ii  libc6:i386                                 2.27-3ubuntu1.4                                                                 i386         GNU C Library: Shared libraries
ii  libc6-dbg:amd64                            2.27-3ubuntu1.4                                                                 amd64        GNU C Library: detached debugging symbols
ii  libc6-dev:amd64                            2.27-3ubuntu1.4                                                                 amd64        GNU C Library: Development Libraries and Header Files
rc  libc6-i386                                 2.27-3ubuntu1                                                                   amd64        GNU C Library: 32-bit shared libraries for AMD64

Probably it's a libc dependency not satisfied, I'll try new setup with more recent os and libraries to find if error still present. At the moment can't update this system manually with a libc custom built with version 2.29 or greater.

AlessandroPerazzetta, 2021-09-07:
With new updated OS (Mint 20 with libc 2.31) seems all work fine (note that vscodium installed is 1.60):

Version: 1.60.0
Commit: e7d7e9a9348e6a8cc8c03f877d39cb72e5dfb1ff
Date: 2021-09-07T00:20:05.363Z
Electron: 13.1.8
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.4.0-84-generic

alessandro@nuc:~$ cat /etc/issue
Linux Mint 20.2 Uma \n \l

alessandro@nuc:~$ uname -a
Linux nuc 5.4.0-84-generic #94-Ubuntu SMP Thu Aug 26 20:27:37 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

alessandro@nuc:~$ dpkg -l |grep libc
ii  klibc-utils                                2.0.7-1ubuntu5                                                                  amd64        small utilities built with klibc for early boot
ii  libc-ares2:amd64                           1.15.0-1ubuntu0.1                                                               amd64        asynchronous name resolver
ii  libc-bin                                   2.31-0ubuntu9.2                                                                 amd64        GNU C Library: Binaries
ii  libc6:amd64                                2.31-0ubuntu9.2                                                                 amd64        GNU C Library: Shared libraries
ii  libc6-dbg:amd64                            2.31-0ubuntu9.2                                                                 amd64        GNU C Library: detached debugging symbols

Also updated vscodium to 1.60 on previous system (mint 19.3 libc 2.27)  and problem with serial persist.

gcampbell-msft, 2023-02-16:
@AlessandroPerazzetta We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue or if the issue still persists!

AlessandroPerazzetta, 2023-02-18:
Current arduino extension release v.0.5.0 seems working fine on this environment:

alessandro@T400:~$ uname -a
Linux T400 5.0.0-32-generic #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

alessandro@T400:~$ cat /etc/issue
Linux Mint 19.3 Tricia \n \l

VSCodium
Version: 1.75.1
Release: 23040
Commit: c6a9727b801542499d2b59cb7db4224e060fc578
Date: 2023-02-09T22:41:50.017Z
Electron: 19.1.9
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 5.0.0-32-generic
Sandboxed: No

Arduino extension v.0.5.0

alessandro@T400:~$ dpkg -l |grep libc
ii  klibc-utils                                2.0.4-9ubuntu2.2                                                                amd64        small utilities built with klibc for early boot
ii  libc-bin                                   2.27-3ubuntu1.6                                                                 amd64        GNU C Library: Binaries
ii  libc-dev-bin                               2.27-3ubuntu1.6                                                                 amd64        GNU C Library: Development binaries
ii  libc6:amd64                                2.27-3ubuntu1.6                                                                 amd64        GNU C Library: Shared libraries
ii  libc6-dbg:amd64                            2.27-3ubuntu1.6                                                                 amd64        GNU C Library: detached debugging symbols
ii  libc6-dev:amd64                            2.27-3ubuntu1.6                                                                 amd64        GNU C Library: Development Libraries and Header Files

gcampbell-msft, 2023-02-21:
@AlessandroPerazzetta Great! If I'm understanding your comment, it's working correctly, this is great! I'll close this issue, but please feel free to re-open it if I misunderstood your comment. Thanks!