microsoft/vscode-arduino #1339: Serial port does not send data

iqapps, 2021-09-03:
With Arduino Extension 0.44 and VS Code 1.60, the opened serial port does not send out data, when pressing CTRL+ALT+S then typing some characters and pressing ENTER.

Downgrading to Arduino Extension 0.43 and VS Code to 1.58.2, the serial port works as expected.

mustaqmasterstudent, 2021-09-26:
I have closed my issue #1340. Because it is somewhat similar to this issue #1339. But I tried downgrading as mentioned in this issue, but this did not helped me.

masynthetic, 2021-10-30:
having the same issue here, rolling back does not work as old versions do not even allow me to select a serial port. Certain the issue is not related to EOL sequence. 

the fun never ends with this extension...

iqapps, 2021-11-08:
I have not seen this for some time after upgrading to 0.4.6.

closed 2021-11-08 by iqapps