microsoft/vscode-arduino #1340: Serial Port dose not send data.

mustaqmasterstudent, 2021-09-06:
I want to read the string input from the user in serial port.

When I use "Arduino: Sent Text to Serial Port". I was not able to embed the text to serial port even after hitting the enter.

hlovdal, 2021-09-16:
This seems to be a duplicate of #1339, can you verify and check if downgrading like mentioned in that issue makes any difference?

mustaqmasterstudent, 2021-09-16:
Yes, I already tried that. 

I tried after downgrading, but it was not even compiling for me. So, I upgraded again. 

Which is also similar issue of #1339 and #1347.

hlovdal, 2021-09-21:
So this issue can be closed as a duplicate then, and all effort be put into #1339 I assume?

mustaqmasterstudent, 2021-09-26:
Okay! I close my issue, as this is just duplicate of #1339 issue.

closed 2021-09-26 by mustaqmasterstudent