microsoft/vscode-arduino #1341: Option to add costum libraries to project folder

Yorams, 2021-09-06:
I want to have a .libraries folder with my own costum libraries. Right now the compiler cannot find the libraries, even if they are in the includePath of c_cpp_properties.json

The option talked about in this issue is what i mean: https://github.com/microsoft/vscode-arduino/issues/830. But is is closed by a bot. The only way to add costum libraries is to copy them in the default libraries folder. And that is not a good workflow.

barreiroleo, 2021-09-10:
In IDE versions higher than 1.6 if you want to include libraries from the project folder, you need to put them in a src subfolder. For example:
`#include "src/YourLib/YourLib.h"`

sahlex, 2022-09-08:
This does not solve the problem, barreiroleo. As suggested in the online docs of arduino-cli (https://arduino.github.io/arduino-cli/0.27/sketch-build-process/), any library contained in a subfolder ```src``` should be referencable. But if this library (e.g. ArduinoBearSSL) contains subfolders that have headers referenced with ```<bearssl/bearssl_hash.h>``` the build breaks nevertheless. So this approach works for self-contained files but not for libraries having a complex directory structure (like ArduinoBearSSL does).

However, there is a workaround: putting the libraries in a folder named ```libs``` (for instance), and writing a tasks.json that passes the libs directory with ```--libraries``` as an argument to arduino-cli works as intended. But this is cumbersome because the 'Verify'-button is no longer usable, only the task will do.

barreiroleo, 2022-09-08:
It depends on your approach. The `Libs` folder is generally used for third party packages. Placing your own files in the `src` folder gives you the ability to work on your packages without rewriting `Cflags`. I've been experiencing issues with the vscode extension overwriting my declarations. I no longer work with vscode and neither with Arduino framework, I don't know about the state of the art today.
Cheers