microsoft/vscode-arduino #1343: Cannot select the Arduino configuration for `.vscode/launch.json` when the `.ino` file is selected

zfields, 2021-09-08:
This took me forever to figure out...

My typical workflow has the `.ino` file open when I want to setup the debugger.

Luckily, when I was first playing with the extension, I clicked through all the menus to see how it worked and knew `Arduino` was an option. So I just kept clicking and digging until I figured out the pattern to make it appear again.

In the repro below, I selected the `BareMinimum.ino` sketch from the `Arduino: Examples` in the F1 menu. Then I started recording. You can see the Arduino option exists with the welcome screen open, but as soon as I select the `.ino` file, then Arduino no longer appears as an option. If I close the `.ino` file, then Arduino reappears as an option.

[image]

The behavior I am expecting is that Arduino would always be an option, especially when an `.ino` file is in the foreground.

zfields, 2022-06-01:
@benmcmorran I just repro'd this while helping a friend today. It's been several months, since I first reported, can someone look over this?

fredsa, 2023-08-08:
Indeed. Likely same issue as reported in #1111.