microsoft/vscode-arduino #1346: Unable to find "Arduino IDE" when option "arduino-cli" is set to true under Linux

legerborea, 2021-09-09:
Hello, I have an issue under Ubuntu 20.04 when setting "Use Arduino Cli" (I am using this option under Win10 also and this work properly).

First, my `settings.json` :
```json
"arduino.useArduinoCli": true,
"arduino.path": "/snap/bin/arduino-cli",
"arduino.additionalUrls": ""
```
> `arduino-cli` was installed using snap store (https://snapcraft.io/arduino-cli)

Usage of `arduino-cli` is set to true, the path is also valid but when I try using arduino extension, I get error message _"Cannot find Arduino IDE. Please specify the 'arduino.path'"_.  
Why extension search for Arduino IDE ? Did I miss something ?
I search for similar isssue ([#950](https://github.com/microsoft/vscode-arduino/issues/950)) but solution are for Windows OS.

Thank you for your time

Galex1223, 2021-09-14:
I have the same problem @legerborea , were you able to fix it or do something ?

legerborea, 2021-09-15:
Hello @Galex1223, no I try to set the **commandPath** to `arduino-cli` (instead of `arduino` default value). But I ended up with an other error :
```
[Starting] Update package index files...
dummy
Error initializing instance: Error downloading library_index.json.gz: Get "https://downloads.arduino.cc/libraries/library_index.json.gz": dial tcp: lookup downloads.arduino.cc on [::1]:53: dial udp [::1]:53: socket: permission denied
Invalid argument passed: invalid item dummy
[Error] Exit with code=7
```
So I am not sure that setting **commandPath** value is enough...

pruge, 2021-11-22:
try this

```JSON
"arduino.useArduinoCli": true,
"arduino.path": "/snap/bin/",
"arduino.commandPath":"arduino-cli",
"arduino.additionalUrls": ""
```

nick-pestell, 2021-12-16:
I am experiencing the same issue. I have tried both settings.json configurations suggested by @legerborea  and @pruge. In both cases I receive the error message "_Cannot find Arduino IDE. Please specify the 'arduino.path'"_.

I have confirmed the instillation of arduino-cli is working and I am pointing to the correct fie.

Galex1223, 2021-12-17:
> try this
> 
> ```json
> "arduino.useArduinoCli": true,
> "arduino.path": "/snap/bin/",
> "arduino.commandPath":"arduino-cli",
> "arduino.additionalUrls": ""
> ```

Hey @pruge. sorry for the delay! My arduino-cli installation is not located in /snap/bin, but I pointed to it correctly using the same other setting as you and still has the error messagge refered by @nick-pestell  ! 

Thank you for your update.

pruge, 2021-12-18:
1. install arduino-cli
```bash
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh
```

2. check arduino version
```bash
$ /Users/<user id>/bin/arduino-cli version
```

```bash
arduino-cli  Version: 0.20.2 Commit: 13783819 Date: 2021-12-09T13:32:05Z
```

3. vscode arduino options
``` json
  "arduino.path": "/Users/<user id>/bin/",
  "arduino.commandPath": "arduino-cli",
  "arduino.useArduinoCli": true,
  "arduino.defaultBaudRate": 9600,
```

RanzQ, 2021-12-18:
I was able to get the normal snap arduino working with these paths: [#1163 ](https://github.com/microsoft/vscode-arduino/issues/1163#issuecomment-997155241)

Maybe similar paths work for arduino-cli.

nick-pestell, 2021-12-18:
Thank you @pruge, I seem to have it working now.

kopp, 2022-12-30:
I added https://github.com/microsoft/vscode-arduino/pull/1562 to document this. @legerborea : I guess we can close the issue, right?

legerborea, 2023-01-01:
Hello @kopp,
Thank you for the review. Unfortunately, I have never succeeded to use this extension with _arduino-cli_ from **snap**. 

I solved message error: _"Cannot find Arduino IDE. Please specify the 'arduino.path"_ with those parameters in extension parameters:
```json
"arduino.useArduinoCli": true,
"arduino.path": "/snap/bin/",
"arduino.commandPath": "arduino-cli"
```

But then, I am unable to list boards or upload sketches... So I have dropped snap version of `arduino-cli` and use official binary instead, which perfectly works with this extension.

Long story short, issue can be closed 😉

closed 2023-01-01 by legerborea