microsoft/vscode-arduino #1347: Help!      Send Text to Serial Port-No response

sasakihin, 2021-09-09:
Use Arduino IDE is ok
but use vscode Send Text to Serial is no response

vscod 1.60
vscode-arduino 0.4.4
Arduino 1.8.15
win10 20H2
Board: widora-air(esp32)

hlovdal, 2021-09-16:
This seems to be a duplicate of #1339, can you verify and check if downgrading like mentioned in that issue makes any difference?

benmcmorran, 2022-01-10:
Are you still seeing this issue on the latest version 0.4.8 of the extension? I can't reproduce this issue with the code below and I suspect there were changes in the [serial-monitor-cli project](https://github.com/microsoft/serial-monitor-cli) to fix it.

```c
void loop()
{
    auto str = Serial.readString();
    Serial.println(str);
    delay(1000);
}
```

closed 2022-02-18 by github-actions[bot]