microsoft/vscode-arduino #1348: Serial port console show only complete line

vlastahajek, 2021-09-12:
Since the v0.4.4 (on Windows 10) serial port console seems to wait for LF to print new data.

When printing, for example, WiFi connection progress by char by char via `Serial.print()`, there is nothing shown until `Serial.println()`.
Also, when using ESP8266 with serial speed 74880  on SDK  3.0.2, when it reboots, it sends boot info with speed 115200, and at this time Serial console stops.

vlastahajek, 2021-09-22:
Both problems still exist on 0.4.5

apws, 2022-01-19:
thanks, its still issue, I used only print(".") initially and as nothing was happening, was hard to track for a while whats wrong where ... please, fix this :-)

binarynoise, 2022-01-25:
Same issue here. Using `Serial.print(".");` to visualize progress and wondered why there's nothing on the serial monitor. And I don't think, it would be a good idea to just `println()` everything

benmcmorran, 2022-02-09:
This should be fixed by #1450, which will be included in the next release of the extension.

apws, 2022-02-09:
luckily, this feature led me to termite monitor/terminal where I am more than happy for my purpose ))

benmcmorran, 2022-02-11:
This issue has been fixed in the latest release of this extension, which is available in the VS Code extension marketplace.

closed 2022-02-11 by benmcmorran

vlastahajek, 2022-02-21:
I can confirm that. Thanks!