microsoft/vscode-arduino #1354: Intellisense Error for Raspberry Pi Pico boards

sstaub, 2021-09-23:
Here a screenshot when using Raspberry Pi Pico boards, get an error which shouldn't happen
[image]

steeley, 2022-01-26:
I have exactly same error, and have spent hours trying to fix it. Nothing works!!
Arduino IDE works fine.

sstaub, 2022-01-26:
Maybe it is a general problem 3rd party cores

StefanKellerAC, 2022-12-10:
The extension does not find the paths in the core's subdirectory  \Pico-SDK

that may be a problem with \ and / when the extension parses the output, which is e.g.:
```
"-IC:\\Users\\stefan\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.4/pico-sdk/src/rp2040/hardware_regs/include/"
```
they are included by
" \rp2040\hardware\rp2040\2.6.4\lib\platform_inc.txt"
the "standard" includes are found

I proposed a dirty fix there:
https://github.com/earlephilhower/arduino-pico/issues/960#issuecomment-1345235920

tylerjwatson, 2023-05-13:
Hi all,

I've created a PR for the `cocopa` library that adds support for [GCC Directory Options]() that is required for arduino-pico to work in this extension.

[image]

is now 

[image]

and my project now works properly.

This library hasn't been updated in a good while, but I hope the patch is merged in and dependency updated. My rp2040 project does not work correctly without it.

tylerjwatson, 2023-05-14:
in the mean time, you're welcome to try out my build here: https://github.com/tylerjwatson/vscode-arduino/releases/tag/v0.6.1