microsoft/vscode-arduino #1360: v0.4.6 board selection does not show any board

protttoy, 2021-09-30:
tried both on vscode 1.58 and 1.60.2

reverting back to v0.4.3 works, bur serial selection is not working with vscode 1.60.2

vscode 1.60.2 with arduino-vscode v0.4.5 works fine

System : windows 10 build 19043

hdpklm, 2021-09-30:
I have same problem

mcedrdiego, 2021-10-01:
Same here, Select board list is empty.

philipjoep, 2021-10-12:
I have the same issue: v0.4.6
and
[image]

yaniot, 2021-10-12:
> 我有同样的问题：v0.4.6 和 [image]

I suggest you to roll back the arduino plugin to 0.4.5 vscode I choose 1.58

benmcmorran, 2022-01-05:
This should be fixed in versions 0.4.7 and later by #1367

PR #1367 (2021-11-05): Fix empty board and examples list
This project is still running on react 15, and these `UNSAFE_*`-lifecycle aliases where not introduced until 16.3, hence they are never called.

fixes #1163, fixes #1360, fixes #1363, fixes #1361, fixes #1372
closed 2022-01-05 by benmcmorran