microsoft/vscode-arduino #1361: Bug of vscode-arduino: no results found when selecting board

wxutopia, 2021-09-30:
I have configured vscode-arduino extension as requested. I set "arduino.path" to "C:\Program Files (x86)\Arduino", and set "arduino.commandPath" to "arduino_debug.exe". But when I open blink.ino and click the <Select Board Type>, it says "No results found". I am pretty sure that my board (Arduino Uno) has been connected to my computer and I can upload my program through Arduino IDE. How can I fix the problem?

[image]

[image]

Arduino IDE version: 1.8.16
Arduino IDE installation path: C:\Program Files (x86)\Arduino
vscode version: 1.60.2
OS: windows 10

84ace, 2021-09-30:
I too have this issue now, have been running fine for months.
vscode: 160.2
OS: OSX
Arduino: 1.8.14

fredr, 2021-10-01:
duplicate of #1360, reverting back to 0.4.5 solved the problem for me

NyxOne, 2021-10-01:
> duplicate of #1360, reverting back to 0.4.5 solved the problem for me

Works indeed! Implies that this issue was introduced in 0.4.6!

wxutopia, 2021-10-06:
Problem solved by reverting vscode-arduino version to 0.4.5.

PR #1367 (2021-11-05): Fix empty board and examples list
This project is still running on react 15, and these `UNSAFE_*`-lifecycle aliases where not introduced until 16.3, hence they are never called.

fixes #1163, fixes #1360, fixes #1363, fixes #1361, fixes #1372