microsoft/vscode-arduino #1373: Failed to generate IntelliSense configuration with esp32 2.0.0

atyshka, 2021-10-31:
When upgrading my esp32 package to version 2.0.0, I get the message "Failed to generate IntelliSense configuration". Downgrading to 1.0.6 fixes the issue, but I want to take advantage of the new hardware serial features. Any ideas what could be causing this? Is there a way to get a more detailed description of the failure?

WarpedPixel, 2021-10-31:
I see exactly the same issue. Create a new project with Arduino: Initialize, select any esp32 platform (from the 2.0.0 board install), and it will compile projects then print `[Warning] Failed to generate IntelliSense configuration.`. It also does not generate an Arduino Intellisense profile. Select a different board (like an Arduino Duo) and it will not print the error.

rev1204, 2021-11-19:
Same here, I'm using ESP32 core version 2.0.1. Any fix/workaround to this problem?

rev1204, 2021-11-19:
Ok, so weird thing I notice is sometime C/C++ show warning:
`IntelliSenseMode was changed from "windows-gcc-x64" to "windows-gcc-x86"`
So this problem seems to happen when Visual Studio (not the code, the IDE one) with MSVC installed. Apparently something cause confusion to the C/C++ extension. This is just a workaround, but deleting (or rename) `cl.exe` inside
`C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\Hostx64\x64`
solve the problem. Still, I'm waiting a proper fix to this problem.

ankit-thealchemist, 2022-01-01:
Me also having the same issue...

mahyar1284, 2022-01-23:
same issue!!! :(

enwi, 2022-01-24:
I don't know if this is the case for you, but I had this issue whenever the `.ino` file was not named exacctly after the folder it was in.
So for example if your `.ino` file is called `test.ino` make sure it is inside a folder called `test`.

pruge, 2022-01-30:
same issue!!!

martinlong1978, 2022-02-07:
I was having this issue. I checked the configuration being used in the bottom right of the status bar, and it had been set back to Win32. I set this back to Arduino and intellisense is now working fine. I still get the message sometimes, but it seems to have resolved all of the squigglies, and intellisense is working properly now.

codysorgenfrey1, 2022-02-16:
Same problem here on both MacOS and Windows 10. Esp8266 boards work fine, ESP32 boards fail to generate intellisense configuration.

Llaves, 2022-03-08:
I have the same issue, but my target board is a Wemos D1 mini, which is an esp8266, not esp32. This occurs only for some projects, not others. I haven't been able to track down the differences that allow one project to generate intellisense, but not the other

siktec-lab, 2022-03-29:
> I don't know if this is the case for you, but I had this issue whenever the `.ino` file was not named exactly after the folder it was in. So for example if your `.ino` file is called `test.ino` make sure it is inside a folder called `test`.

That's actually the solution - For some reason the Arduino get into trouble when the main.ino is not called sketch.ino or has the same name as its directory.

Medeirox, 2022-06-27:
using the same project_filename.ino name and the folder name has solved the problem for me too. I'm not sure, but I think I did just like @Llaves did... renamed the directory and file, and then restarted VSCode.

itopaloglu83, 2022-08-26:
MacOS users should also make sure that both VSCode and Arduino app have the necessary disk rights to access where the files are located. In my case, I was experiencing this issue because VSCode was not allowed to have access to my downloads folder. 

You find the disk access settings under System Preferences > Security & Privacy > Privacy Tab > Files and Folders.