microsoft/vscode-arduino #1374: Build process should start in the sketch directory

AlexIII, 2021-11-03:
## Description

Building with `arduino-builder.exe` should start in the sketch directory in order to comply to Arduino IDE. 
Right now it starts from the extension's some sub-directory.
 
This leads to problems when including files with relative path from inline assembler, because relative path in inline assembler is relative to the current directory in which the compiler is running (and not to the source file).

For example, it makes it impossible to use [incbin](https://github.com/graphitemaster/incbin) which works fine in the original Arduino IDE.

## Test case

`TEST1/test.bin` - empty file

`TEST1/sketch1/sketch1.ino`

```
__asm__ (
    ".incbin \"../test.bin\"\n"
);
void setup() {}
void loop() {}
```

This sketch will compile with Arduino IDE, but not with  vscode-arduino.

AlexIII, 2021-11-03:
PR provided

benmcmorran, 2021-11-05:
PR merged.

PR #1375 (2021-11-05): Fix build cwd (#1374)
fixes #1374
closed 2021-11-05 by benmcmorran