microsoft/vscode-arduino #1388: ArduinoCLI additional urls fails

albydnc, 2021-11-19:
When passing the additionalUrls setting, the board manager can't update and the terminal says:
```

[Starting] Update package index files...
dummy
Invalid argument passed: invalid item dummy
[Error] Exit with code=7
```
Tried with a single string and in an array, same results.
My settings.json is:
```json
{
    "arduino.path": "/home/albertoperro/bin/",
    "arduino.commandPath": "arduino-cli",
    "arduino.useArduinoCli": true,
    "arduino.additionalUrls": "https://dl.espressif.com/dl/package_esp32_index.json"
}
```

VSCode version `1.62.3 ccbaa2d27e38e5afa3e5c21c1c7bef4657064247 x64`
Arduino extension version `v0.4.8 preview`
Arduino CLI version `arduino-cli alpha Version: 0.19.3 Commit: 12f1afc2 Date: 2021-10-12T10:15:19Z`

joybanerjee08, 2021-11-21:
I had this same issue. 
I think [this was the line](https://github.com/microsoft/vscode-arduino/blob/6a1731747a762f02532d4ec372973db00b0b096a/src/arduino/arduino.ts#L99) which was causing the error in my case.
Installing Arduino IDE and removing CLI, then modifying the path to IDE fixed my problem.

phan-t, 2022-01-12:
I had the same error output:

```
[Starting] Update package index files...
dummy
Invalid argument passed: invalid item dummy
[Error] Exit with code=7
```

This is what I did to workaround the issue: 

1. Added the additional urls directly to `arduino-cli` config by executing:

```
arduino-cli config init --additional-urls https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
```
2. Executed `arduino-cli core update-index`. You'll see the following output:

```
Updating index: package_index.json downloaded                                                                                                                   
Updating index: package_index.json.sig downloaded                                                                                                               
Updating index: package_esp32_index.json downloaded
```

After performing the above, I could view the `esp32` package in in the Arduino Board Manager and install it successfully.

Hope this helps.

isharris, 2022-02-18:
Thanks @phan-t worked like a charm on ubuntu

strazto, 2022-10-22:
https://github.com/microsoft/vscode-arduino/issues/1388#issuecomment-974882923 

> I think [this was the line](https://github.com/microsoft/vscode-arduino/blob/6a1731747a762f02532d4ec372973db00b0b096a/src/arduino/arduino.ts#L99) which was causing the error in my case.

I think you're right about that, I quickly searched the extension's code and that line hasn't been touched in 6 years and specifically mentions the IDE.

> Installing Arduino IDE and removing CLI, then modifying the path to IDE fixed my problem.

Although I can see how that would get around it, it's probably not the ideal workaround, especially due to https://github.com/microsoft/vscode-arduino/issues/1477 currently requiring the CLI as a workaround if 2.0 is installed

https://github.com/microsoft/vscode-arduino/issues/1388#issuecomment-1010547815 is probably the best, and I might try submitting a PR to fix later

brunoga, 2023-02-18:
Any updates on this? I am trying to use the bundled arduino-cli so the suggestion here does not work (I mean, I do not know what is the path to use for arduino-cli in that case).

MatthewSteeves, 2023-02-18:
Re: location of bundled arduino-cli, if on Windows, check under:
`C:\Users\YOUR_USERNAME\.vscode\extensions\vsciot-vscode.vscode-arduino-0.5.0-win32-x64\assets\platform\win32-x64\arduino-cli`

IPlayZed, 2023-03-15:
So basically, you guys want to move to the bundled arduino-cli in the future, and it does not work if one adds custom boards... What a great idea.

benmcmorran, 2023-03-15:
Can you try switching to the pre-release version? It includes #1611 which should fix this issue. I'll try to get another official release out soon.

[image]

benmcmorran, 2023-03-15:
I just published non-pre-release version 0.6.0 to the marketplace as well, which also includes #1611.

BrettThePark, 2023-11-29:
This appears to still be broken as of v0.6.230727001

SergioGCarvalho, 2024-02-11:
if using mac don't forget ide permissions https://github.com/microsoft/vscode-arduino/issues/1681#issuecomment-1747890450

Anaphase, 2024-03-15:
@SergioGCarvalho thank you so much, that did the trick for me!