microsoft/vscode-arduino #1389: Arduino (Not CLI), Can't include header file in root of project

thewh1teagle, 2021-11-20:
Hello
I try to include simple header file in the root of the project and upload it to the arduino using vscode. (`ESP8266`)
I have two files in the project

<details>
  <summary>src/main.ino</summary>
  
```c
#include "test.h"
void setup(){}
void loop(){}
```

</details>

<details>
  <summary>src/test.h</summary>
  
```c
define TEST "test"
```
</details>

When I try to verify it (using vscode command)
[image]

I get this
<details>
  <summary>error</summary>
 
```bash
Loading configuration...
Initialising packages...
Preparing boards...
Verifying...
main:12:10: fatal error: test.h: No such file or directory
1 | #include "test.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
```
</details>

<details>
  <summary>.vscode/settings.json</summary>

```json
{
      "arduino.path": "C:\\Users\\User\\Downloads\\arduino-1.8.16-windows\\arduino-1.8.16",
      "arduino.commandPath": "arduino.exe",
      "C_Cpp.default.browse.path": [
          "C:\\Users\\User\\Documents\\Arduino\\libraries",
          "C:\\Users\\User\\Downloads\\arduino-1.8.16-windows\\arduino-1.8.16\\libraries\\**",
          "C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\**",
          "${workspaceFolder}"
      ],
      "C_Cpp.default.includePath": ["C:\\Users\\User\\Desktop\\arduino_app\\src\\**"],
      "C_Cpp.default.browse.databaseFilename": "",
      "C_Cpp.intelliSenseEngine": "Tag Parser",
      "C_Cpp.intelliSenseEngineFallback": "Disabled",
    "C_Cpp.default.compilerArgs": [
        "-I ${workspaceFolder}\\src"
    ],
    "C_Cpp.default.intelliSenseMode": "msvc-x64",
    "C_Cpp.default.browse.limitSymbolsToIncludedHeaders": false,
    "C_Cpp.default.defines": ["ARDUINO=100"],
    "C_Cpp.default.forcedInclude": ["C:\\Users\\User\\Desktop\\arduino_app\\src\\test.h"]   
}
```
</details>

<details>
  <summary>.vscode/arduino.json</summary>

```json
{
    "configuration": "xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200",
    "board": "esp8266:esp8266:generic",
    "sketch": "src\\main.ino",
    "port": "COM3",
    "output": "src\\dist"
}
```
</details>

platform: `windows 11 x64`
extension version: `0.4.8`
arduino ide version`1.8.16`
vscode version `1.62.3`

hdpklm, 2021-11-22:
you can move the test.h to a sub folder like src/test/test.h
and then change the include to the new path.
its work for me.

and i think you have to modify the test.h file in definition line to #define TEST "test"

thewh1teagle, 2021-11-22:
@hdpklm 
The `test.h` file was just an example. I edited this.
This is Slightly crooked solution, but better than nothing :)
Can you share your config files with your directory structure?

hdpklm, 2021-11-22:
For me i think is better for reorder all directory when I start a project 

but I have a quiestion. How you add the ocardion in github ??
[image]

```
{
	"board": "esp32:esp32:esp32",
	"port": "COM4",
	"sketch": "main.ino",
	"output": "0-build",
	"configuration": "PSRAM=disabled,PartitionScheme=min_spiffs,CPUFreq=240,FlashMode=dio,FlashFreq=80,FlashSize=2M,UploadSpeed=921600,DebugLevel=none"
}

```

```
{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0\\bin\\xtensa-esp32-elf-g++",
            "compilerArgs": [
                "-std=gnu++11",
                "-Wpointer-arith",
                "-fexceptions",
                "-fstack-protector",
                "-ffunction-sections",
                "-fdata-sections",
                "-fstrict-volatile-bitfields",
                "-mlongcalls",
                "-nostdlib",
                "-Wno-error=maybe-uninitialized",
                "-Wno-error=unused-function",
                "-Wno-error=unused-but-set-variable",
                "-Wno-error=unused-variable",
                "-Wno-error=deprecated-declarations",
                "-Wno-unused-parameter",
                "-Wno-unused-but-set-parameter",
                "-Wno-missing-field-initializers",
                "-Wno-sign-compare",
                "-fno-rtti"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\config",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\app_trace",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\app_update",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\asio",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\bootloader_support",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\bt",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\coap",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\console",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\driver",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\efuse",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp-tls",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp32",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp_adc_cal",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp_event",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp_http_client",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp_http_server",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp_https_ota",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp_https_server",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp_ringbuf",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\esp_websocket_client",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\espcoredump",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\ethernet",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\expat",
                "C:\\Users\\Hassan Daoud\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\sdk\\include\\fatfs",
[... 282 lines cut ...]
                "__INT16_C(c)=c",
                "__STDC__=1",
                "__PTRDIFF_TYPE__=int",
                "__ATOMIC_SEQ_CST=5",
                "__UINT32_TYPE__=long unsigned int",
                "__UINTPTR_TYPE__=unsigned int",
                "__DEC64_SUBNORMAL_MIN__=0.000000000000001E-383DD",
                "__DEC128_MANT_DIG__=34",
                "__LDBL_MIN_10_EXP__=(-307)",
                "__SIZEOF_LONG_LONG__=8",
                "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
                "__LDBL_DIG__=15",
                "__FLT_DECIMAL_DIG__=9",
                "__UINT_FAST16_MAX__=0xffffffffU",
                "__GNUC_GNU_INLINE__=1",
                "__GCC_ATOMIC_SHORT_LOCK_FREE=2",
                "__UINT_FAST8_TYPE__=unsigned int",
                "__ATOMIC_ACQ_REL=4",
                "__ATOMIC_RELEASE=3",
                "USBCON"
            ]
        }
    ]
}
```

thewh1teagle, 2021-11-22:
~~~markdown
<details>
  <summary>src/main.ino</summary>
 
```c
#include "test.h"
void setup(){}
void loop(){}
```
</details>
~~~

This 👆, make this 👇

<details>
  <summary>src/main.ino</summary>
 
```c
#include "test.h"
void setup(){}
void loop(){}
```
</details>

hdpklm, 2021-11-22:
@thewh1teagle 

ok thank you so mutch

doyleLai, 2021-11-23:
> 
It is so stupid that herder files should be put in a sub-folder in order to make it works.  
I am facing the same problem in VS code but my project works well with Arduino ide 1.8 and 2.0.
This bug (or feature?) makes this extension entirely useless for me.

kopp, 2022-12-30:
If this is a known restriction, mentioning it in [the Readme](https://github.com/microsoft/vscode-arduino/blob/main/README.md) would be a first step.  I opened https://github.com/microsoft/vscode-arduino/pull/1561 for that.  It has cost me some time to find this solution, and it nearly discouraged me to use this extension.

jleaders, 2023-07-23:
Is there no way to handle my libraries? They have deeply nested subdirectories that assume proper IncludePaths?

I.e., in lib/mylibary/somefile.h
assumes "lib/mylibary/" is in the include path so it does:
`#include "somefile2.h"`

this now fails?

joecorsi, 2023-10-07:
this is a crazy issue. using libraries is essential to arduino. when will this be fixed?

joecorsi, 2023-10-07:
im having i think a related issue, when i try to include a adafruit library i know is installed, which i installed through vs code, it still cant find it. does this issue apply to installed libraries as well?