microsoft/vscode-arduino #1391: "Exit with code=1" when uploading sketch

nabelekt, 2021-11-26:
FIrst of all: Is there a better place for questions like this? Many Github projects use the Discussions feature now, which I think may be better suited for issues like this, until it's clear that it's a bug.

I am working on transitioning my [DxCore](https://github.com/spenceKonde/dxCore/) project from the Arduino IDE to VS Code. I am able to upload my sketch just fine in the Arduino IDE using either the DxCore:jtag2updi or DxCore:serialupdi57k programmers, but if I try to do the same in VS Code, I see
```
...
Global variables use 4 bytes (0%) of dynamic memory, leaving 16380 bytes for local variables. Maximum is 16384 bytes.
Uploading...
An error occurred while uploading the sketch
...
[Error] Uploading sketch 'test_1_LEDs\test_1_LEDs.ino': Exit with code=1
...
```
in the console output. Full console output is here: [upload_errors_jtag2updi.txt](https://github.com/microsoft/vscode-arduino/files/7610425/upload_errors_jtag2updi.txt), [upload_errors_serialupdi57k.txt](https://github.com/microsoft/vscode-arduino/files/7610426/upload_errors_serialupdi57k.txt). The two are nearly identical.

I'm not seeing anything else particularly useful in the console output. Any ideas on how to resolve this? Any suggestions are appreciated.

Thanks!

thaumatichthys, 2023-01-30:
I am having the same issue

IOT4566, 2024-07-11:
I am having the same issue too!