microsoft/vscode-arduino #1392: Enable all warnings

fakuivan, 2021-12-04:
How can ``-Wall`` and ``-Wextra`` be added so that the verify step shows these warnings in the "problems" panel in vs code? I tried setting ``buildPreferences`` to this but it doesn't seem to have any effect on the output:

```json
    "buildPreferences": [
        ["build.extra_flags", "-Wall"],
        ["compiler.cpp.extra_flags", "-Wall"]
    ]
```

AlexIII, 2021-12-04:
Tried in 0.4.8

Before:
```
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10816 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_597155\\sketch\\Blink.ino.cpp" -o "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_597155\\sketch\\Blink.ino.cpp.o"
```

After I add
```
"buildPreferences": [
        ["build.extra_flags", "-Wall"]
    ]
```

```
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10816 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -Wall "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_308987\\sketch\\Blink.ino.cpp" -o "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_308987\\sketch\\Blink.ino.cpp.o"
```

You can find that the second output has `-Wall` and the first does not.

fakuivan, 2021-12-04:
@AlexIII How do you actually compile the sketch? The only option I get is to "Verify" it.

AlexIII, 2021-12-04:
Verify runs compilation. In order to see full compile process output create `.vscode\settings.json` in your workspace directory and add
```
{
    "arduino.logLevel": "verbose"
}
```

fakuivan, 2021-12-04:
Sure, that shows the warnings, but they don't show up in the problem pane

[image]

[image]

benmcmorran, 2022-01-19:
Since it looks like you got this working in the output pane, I'm going to go ahead and close this issue as a duplicate of #1376. We'll need to change the extension to either use VS Code's task system or do additional parsing of the compiler output to support the problems pane.

closed 2022-01-19 by benmcmorran