microsoft/vscode-arduino #1393: Serial monitor don't work after a maj

Alexisdxyz, 2021-12-09:
Issue Type: <b>Bug</b>

just update Vs code 1.63

Extension version: 0.4.8
VS Code version: Code 1.63.0 (7db1a2b88f7557e0a43fec75b6ba7e50b3e9f77e, 2021-12-07T06:26:56.179Z)
OS version: Windows_NT x64 10.0.22000
Restricted Mode: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz (12 x 2592)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.82GB (8.64GB free)|
|Process Argv|--crash-reporter-id 01d99435-a1d2-4d13-8345-003ed3e32046 --crash-reporter-id 01d99435-a1d2-4d13-8345-003ed3e32046|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383cf:30185419
vspor879:30202332
vspor708:30202333
vspor363:30204092
pythontb:30283811
pythonptprofiler:30281270
vsdfh931:30280409
vshan820:30294714
vstes263:30335439
vscorecescf:30384386
pythondataviewer:30285071
vscod805cf:30301675
pythonvspyt200:30340761
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
pythonrunftest32:30373476
pythonf5test824:30373475
javagetstartedt:30391933
pythonvspyt187:30373474
vsaa593cf:30376535
pythonvs932cf:30408637
vscop804cf:30404767
vscop453:30404998
vsrem710cf:30405999
vscexrecpromp3t1:30407762

```

</details>

benmcmorran, 2022-01-10:
I'm not able to reproduce this issue with extension 0.4.8 on VS Code 1.63.2 running on Windows. @Alexisdxyz (or upvoters @CodingCreate101, @dhassault, and @cschaffter) are you still seeing this issue, and if you are can you share a minimal code snippet that reproduces the issue? I'm wondering if this is the same problem as #1396 and #1402.

CodingCreate101, 2022-01-10:
I have not checked recently. I will try to get my versions and see if it's the same issue by adding some delay. 

Maybe by weekend 😬

BelKed, 2022-01-15:
Serial Monitor from `vscode-arduino` doesn't work properly in my case. It doesn't show any output. After running "Arduino: Open Serial Monitor" it outputs just:

```
[Starting] Opening the serial port - /dev/cu.SLAB_USBtoUART
```

Seems like it trying to connect to that serial port, but it can't... I tried it with and without opened Serial Monitor in Arduino IDE, but no results (it should show error in VS Code).

Using `screen` command (`screen /dev/cu.SLAB_USBtoUART 115200`) and also in Arduino IDE 1.8.15 it works as it should.

Extension version: 0.4.10
VS Code version: Code 1.63.2 (899d46d82c4c95423fb7e10e68eba52050e30ba3, 2021-12-15T09:37:28.172Z)
OS version: Darwin x64 17.7.0 (macOS High Sierra 1.13.6)
Restricted Mode: No

---

With vscode-arduino 0.4.3 and Code 1.58.2 it works as intended:

```
[Starting] Opening the serial port - /dev/tty.SLAB_USBtoUART
[Info] Opened the serial port - /dev/tty.SLAB_USBtoUART
Hello World!
Hello World!
Hello World!
Hello World!
Hello World!
Hello World!
Hello World!
[Done] Closed the serial port
```

benmcmorran, 2022-01-18:
Thanks for the report @BelKed. Given that version 0.4.3 is the last time the serial monitor worked for you, I'm guessing this issue has something to do with this project's adoption of [serial-monitor-cli](https://github.com/microsoft/serial-monitor-cli), which started in version 0.4.4. Can you try running these commands in the terminal to help narrow down if the problem is coming from serial-monitor-cli itself, or the integration in VS Code?

```bash
~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/main list-ports
```

You should see a JSON response with each port available on your machine. Next, try to connect to one of the ports (I'm using 
`/dev/cu.SLAB_USBtoUART` for the example below).

```bash
~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/main open -b 115200 /dev/cu.SLAB_USBtoUART
```

You should see output from the attached device, assuming it's writing data to the serial port.

BelKed, 2022-01-18:
Yes, you are right, there is some problem with [`serial-monitor-cli`](https://github.com/microsoft/serial-monitor-cli)... It seems like its Python library is built for Mac OS X 10.15 (I have Mac OS X 1.13.6).

Running each of both commands show same error code:
```sh
❯ ~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/main list-ports

[21267] Error loading Python lib '/Users/user/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/libpython3.8.dylib': dlopen: dlopen(/Users/user/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/libpython3.8.dylib, 10): Symbol not found: ____chkstk_darwin
  Referenced from: /Users/user/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/libintl.8.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Users/user/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/libintl.8.dylib
```

benmcmorran, 2022-01-18:
Thanks! That should be enough for us to try to repro the issue.

gcampbell-msft, 2022-01-21:
@BelKed We have a possible solution that we'd like your help testing to confirm it works. Could you possibly follow these steps and let us know if it works?

1. Download the build artifact from this link: [https://github.com/microsoft/serial-monitor-cli/suites/5008861506/artifacts/147099637](https://github.com/microsoft/serial-monitor-cli/suites/5008861506/artifacts/147099637)
2. Unzip the artifact, open the main.zip within it and copy the contents. You should notice that this looks very similar to what is currently located in your system here: "~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/". Replace your system contents with what you just copied. 
3. At this point you should be able to follow the same steps @benmcmorran gave you above to test it and see if it behaves as expected. 

Let us know if this solves your issue!

BelKed, 2022-01-21:
Thanks, @gcampbell-msft! Now it works as intended:

```sh
❯ ~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/main list-ports
[{"port": "/dev/cu.Bluetooth-Incoming-Port", "desc": "n/a", "hwid": "n/a"}, {"port": "/dev/cu.SLAB_USBtoUART", "desc": "CP2102 USB to UART Bridge Controller", "hwid": "USB VID:PID=10C4:EA60 SER=0001 LOCATION=253-1.2"}]
```

```sh
❯ ~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.10/out/serial-monitor-cli/darwin/main open -b 115200 /dev/cu.SLAB_USBtoUART
Hello World!
Hello World!
Hello World!
...
```

Now I can also upload sketches and open Serial Monitor without problems in VSCode:
```
[Starting] Opening the serial port - /dev/tty.SLAB_USBtoUART
Hello World!
Hello World!
Hello World!
...
```

gcampbell-msft, 2022-01-21:
Great! We will work to get this released as soon as we can.

benmcmorran, 2022-01-28:
This issue should be fixed with the insertion of serial-monitor-cli v0.0.7, which will be included in the next release of the vscode-arduino extension.

benmcmorran, 2022-02-11:
This issue has been fixed in the latest release of this extension, which is available in the VS Code extension marketplace.

closed 2022-02-11 by benmcmorran