microsoft/vscode-arduino #1395: Extension Missing from Marketplace

quinn-n, 2021-12-12:
- OS: `Arch Linux`
  - VSCode version: `1.62.0-1`
 
Searching `arduino` in the extensions tab doesn't bring up this extension.

I've tried re-installing vscode and deleting its data directories (`~/.vscode-oss` and `~/.config/Code - OSS/`), this didn't help.

benmcmorran, 2022-01-05:
The extension appears in the Marketplace for me on VS Code 1.63.2. Are you still seeing this issue?

[image]

SpazCode153, 2022-01-11:
Hi benmcmorran, I have the same issue.
OS: Arch Linux(Manjaro Distro)
VSCode version: 1.62.0

Attached is the 3 different searches I have done for the extension.

[image]
[image]
[image]

Is there more information I can provide for this issue?

SpazCode153, 2022-01-11:
Hi benmcmorran,
I found a solution, it seems the software installer for Manjaro installs an older version of visual studio code, if I download visual studio code from the website and use it like that then I can find the extension.

benmcmorran, 2022-01-11:
Thanks for letting me know, and I'm glad you got this working!

closed 2022-01-11 by benmcmorran