microsoft/vscode-arduino #1396: Serial Monitor not updating in real time

BasselMalek, 2021-12-26:
OS: Win10
Versions: Arduino IDE: 1.8.16 || Extension: 0.4.8
Sys info:
VSCode: 1.63.2
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0

Bug:
Whenever using the serial monitor with an Uno it never shows the output in real time, instead it only dumps it all after closing the serial monitor. To make it work i have to increase the delay in program to 300+ ms.

Steps:
1. Upload any code using Serial.println()/print()
2. open serial monitor
3. no output
4. close monitor
5. all output appear

Log file:
[rendererLog.txt](https://github.com/microsoft/vscode-arduino/files/7777010/rendererLog.txt)

siideffect, 2022-01-11:
I have the same issue:

OS = MacOS Catalina 10.15.7
Vscode ⬇️
Version: 1.63.2
Commit: 899d46d82c4c95423fb7e10e68eba52050e30ba3
Date: 2021-12-15T09:37:28.172Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Darwin x64 19.6.0

Arduino extension = v0.4.8

I tried investigating about this.
If i open the serial monitor with arduino ide, i get all the expected output.
If i open the serial monitor with visual studio code, it works correctly only if there is a high delay(1 second).
However, i noticed the serial monitor is working "well": the output console seems to freeze when outputting data, and prints all in blocks of 30 or so...but if i swap, from output to another window(terminal, problems or debug console) and then to output again in the vscode bottom panel, the expected data is there. 
I ve tested this using a counter variable, and i noticed no Serial.println() are actually lost. 
It s "just" a refreshing issue on the output console.

Arduino ide:
[https://www.dropbox.com/s/rx58x1pnu8ieeyl/Schermata%202022-01-11%20alle%2022.21.58.png?dl=0](url)
All working as expected

VS Code:
[https://www.dropbox.com/s/ur6wl5r46towvr9/Schermata%202022-01-11%20alle%2022.23.34.png?dl=0](url)
Opening the serial monitor results in empty console, but if i switch as i wrote above i can see the expected output ⬇️

[https://www.dropbox.com/s/35xcao1ctg5rqx7/Schermata%202022-01-11%20alle%2022.24.42%20%282%29.png?dl=0](url)

Its not a wrong baud rate settings, i ve read almost everything i could read about this issue... i m writing here because i ended up all my options.
Thanks for your attention and your help!

gcampbell-msft, 2022-01-12:
I've been able to repro this and am working on a fix.

closed 2022-01-13 by gcampbell-msft

siideffect, 2022-01-13:
Thank you @gcampbell-msft for your help, but i do have some problem still.
The serial monitor stopped working since the 0.4.9. and now i m not able to see any output in the console.
Using arduino ide, all works as expected. I tried different baud rates, but thats not the problem.

[https://www.dropbox.com/s/yyxzlo0y9ya7m64/Schermata%202022-01-13%20alle%2010.47.45.png?dl=0](url)

theficus, 2022-01-14:
@siideffect Same problem here. I opened bug #1418 to track this.

gcampbell-msft, 2022-01-14:
@siideffect Could you provide some example code of what you're running and also what board you are using? I'm not currently able to repro this issue.

siideffect, 2022-01-14:
> @siideffect Could you provide some example code of what you're running and also what board you are using? I'm not currently able to repro this issue.

Hi @gcampbell-msft , it was a simple gpio reading on a touch pin of a esp32 board,  in the loop() function. 
Nothing so complicated, just a continuous hitting to the Serial monitor with (and without) delay().
Test was developed on a Esp32 dev module board

crackwitz, 2022-06-11:
I'm running a simple sketch that emits a few lines about once or twice a second. the serial monitor seems to have real trouble showing these lines as they happen. there's a delay of upto a second... I'd really rather this thing show me as data comes in, instead of this apparent "polling". if you _have to_ have a polling interval and the default _has to_ be largeish, please make it configurable because I'd wanna dial that down to something more responsive.

dumski, 2022-07-03:
Yes, in version 4.1.2 problem still persists...

crackwitz, 2022-07-05:
Windows 10, random arduino nano, 115200. it's got nothing to do with that because those things work perfectly fine. Putty has no issues displaying the data *instantly*. regular Arduino also has no issues.

I've been dissatisfied with the regular arduino serial monitor since forever, but since I got to use the serial monitor of esp-idf (for esp32 without arduino), my expectations were raised considerably. I expect something close to a proper terminal. having to click a button or menu item just to send a string once (vscode-arduino) is hopefully just a stopgap.

gcampbell-msft, 2022-07-05:
@crackwitz I'll look into the buffering on the new extension, feel free to post an issue about this on the [vscode-serial-monitor issues page](https://github.com/microsoft/vscode-serial-monitor/issues)! 

As for the failing to scroll along, there is an auto scrolling button present 
[image]
in the standalone Serial Monitor that should allow you to keep up with new input.

crackwitz, 2022-07-05:
Ah, so that's what that does! It wouldn't give me a tooltip and I speculated that it means "line wrap". I'll open an issue.

gcampbell-msft, 2022-07-05:
@crackwitz Yes, unfortunately there's a bug that wouldn't show tooltips. It's been fixed and if you switch to the Pre-release version tooltips are restored, but we haven't done an official release of those fixes just yet.

crackwitz, 2022-07-05:
here's a little screen capture showing this extension:
https://user-images.githubusercontent.com/7065108/177406211-2d3142b5-b52b-48f1-a2cf-51a489a6ef91.mp4

for comparison, three other methods: https://github.com/microsoft/vscode-serial-monitor/issues/23