microsoft/vscode-arduino #1398: Arduino Avr device registers undefined

cschaffter, 2021-12-30:
Updated my system  the other day (full reinstall)
VScode  V1.63.2
Arduino Extension 0.4.8   IntelliSense V1.7.1
Arduino IDE 1.8.19
 
IntelliSense fails to resolve  AVR cpu register names and give the following errors:
 "indentifier "TCCR2A" is undefined"   None of the Arduino UNO chip registers apper to be recognized.   Adding  #include <arduino.h> doesn't help.
The code complies and works on the Arduino IDE but fails with VScode  but work before
I updated my development environment.  Been searching a two days now for a solution, nothing I have tried so far has worked.  An uninstall and reinstall didn't help, I am at a loss. 
Any help would greatly be appreciated.   

Thanks Craig

frodomax33, 2022-02-02:
I've just the same issue, it seems like the extension don't define the board type properly to work with the io.h file : 
For example, if i'm using an arduino pro board : "__AVR_ATmega328P__" isn't define in the c_cpp_properties.json file.

bruyeret, 2023-03-16:
The problem comes from the generated `c_cpp_properties.json`.

Go into your build directory, `compile_commands.json` has a compiler argument that should be passed to `c_cpp_properties.json` but is missing.

For me, adding `"-mmcu=atmega328p"` in `"compilerArgs"` did the job.

spinpwr, 2023-10-09:
The solution from @bruyeret does fix this, but only temporarily. 
When I do a compile (upload or verify) the `c_cpp_properties.json` is recreated and the `-mmcu` compiler argument is removed.
So it have to be added again.

bruyeret, 2023-10-09:
I didn't see that you could change the name of the configuration to avoid it being overwritten.
To do that, I checked the "Disable Intelli Sense Auto Gen" in the extension settings, but your solution is a lot cleaner.
Thanks for sharing it!