microsoft/vscode-arduino #1410: Serial monitor can't handle large amounts of data

theficus, 2022-01-08:
What I'm finding is when my connected device is sending a lot of data, the serial monitor seems to stop processing the data and shows nothing. This doesn't happen when using the Arduino app's serial monitor.

theficus, 2022-01-08:
I believe this is actually the exact same issue reported in #1396 and could also be the same as #1402.

theficus, 2022-01-08:
Duplicate

closed 2022-01-08 by theficus

gcampbell-msft, 2022-01-13:
We just released a new version, 0.4.9, that contains a fix for this issue. Let us know if any more issues come up!

theficus, 2022-01-14:
@gcampbell-msft I've upgraded to 0.4.9 and now the serial monitor doesn't work at all. I have to use the Arduino Serial Monitor. When I roll back to the previous version I can at least see some stuff. I'll open a new issue for this.