microsoft/vscode-arduino #1415: Version number missing in release artifact

adiazulay, 2022-01-13:
The version number is missing on the release artifact.
Right now it just shows as vscode-arduino.vsix but should show vscode-arduino-0.4.9-rc1.vsix

benmcmorran, 2022-01-13:
I did that on purpose because it simplified [the ADO build script](https://github.com/microsoft/vscode-arduino/blob/970615a1f10fb23befe3f6ffadd8f1620e2bb954/azure-pipelines.yml#L84). The ADO artifact publishing step doesn't support wildcards. Do you think it will cause a problem to not have the version number in the filename? It's still correct in the `package.json` and on the marketplace.

benmcmorran, 2022-01-18:
Closing for now because most of our downloads are coming from the VS Code marketplace, which does insert the version number into the VSIX name.

closed 2022-01-18 by benmcmorran