microsoft/vscode-arduino #1418: Serial monitor no longer works after upgrading to 0.4.9

theficus, 2022-01-14:
Issue Type: <b>Bug</b>

The serial monitor on 0.4.8 didn't work well, but I could still see some data such as when I reset my microcontroller.

After upgrading to 0.4.9 now I don't get anything at all when using the serial monitor. The only way I can view anything is to use the Arduino Serial Monitor. This appears to be a regression introduced by #1412.

I've confirmed that rolling back to 0.4.8 I am once again able to see some data over the serial monitor.

Extension version: 0.4.9
VS Code version: Code 1.63.2 (899d46d82c4c95423fb7e10e68eba52050e30ba3, 2021-12-15T09:37:28.172Z)
OS version: Darwin x64 20.6.0
Restricted Mode: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz (16 x 2400)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|1, 2, 2|
|Memory (System)|32.00GB (2.15GB free)|
|Process Argv|--crash-reporter-id 4874e46e-bd6d-4247-accf-2cc334df3d8f|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368cf:30146710
vsreu685:30147344
python383cf:30185419
vspor879:30202332
vspor708:30202333
vspor363:30204092
pythontb:30283811
pythonptprofiler:30281270
vsdfh931:30280409
vshan820:30294714
vstes263:30335439
vscoreces:30384385
pythondataviewer:30285071
vscod805:30301674
pythonvspyt200:30340761
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
vsaa593cf:30376535
vsc1dst:30421291
pythonvs932:30410667
vscop804:30404766
vscop940:30404999
vsrem710:30416614
vsccppwtstudy:30421978

```

</details>

gcampbell-msft, 2022-01-14:
Could you provide some example code of what you're running and also what board you are using? I'm not currently able to repro this issue. 

Are you aware of whether this issue is mac specific?

gcampbell-msft, 2022-01-14:
Are those of you who reacted a thumbs-up to the original post also seeing this issue? If so, could you comment your system information and possibly some example code, as well as describe your issue?

benmcmorran, 2022-01-15:
I've tracked this down to an issue with our new VSIX signing process that accidentally removed certain critical serial monitor files from the extension on Mac and Linux. e0f29b8cf79c9bc3649c42cea4b945254a537af9 disables VSIX signing, and I've verified the files are now present again in builds produced after that commit. I'm currently testing on Linux, and if all goes well, I plan to ship a new release of the extension with the fix in an hour or two.

benmcmorran, 2022-01-15:
I hit one other issue where the VSIX no longer marks the serial-monitor-cli files as executable. That should be an easy enough fix, but at this point I don't expect to get a new release out until Monday.

closed 2022-01-15 by github-actions[bot]