microsoft/vscode-arduino #1427: Failed to open serial port (TypeError)

Triangle4, 2022-01-19:
Hey Guys, 

**My Computer**
3Gen Ryzen

**My Arduino:**
Mega 2560

**My Sketch:**
Blink (unchanged from the original code)

I am not sure why I would get this error:

Any ideas as to why I might get this error:
_**Failed to open serial port COM9 due to error: + TypeError: Cannot read property 'then' of undefined**_

benmcmorran, 2022-01-19:
Thanks for the report @Triangle4! To help diagnose this issue, can you post the content of `%USERPROFILE%\.vscode\extensions\vsciot-vscode.vscode-arduino-0.4.10\arduino.log`? That should let us identify which line is causing the crash. Please remove any sensitive information before posting the log.

Triangle4, 2022-01-19:
Added it to the initial post. Its in the last dropdown

benmcmorran, 2022-01-19:
Thanks. The error line that corresponds to the message you were seeing is `vs/workbench/services/extensions/node/extensionHostProcess.js:85:821)` which is surprising because that code is actually part of VS Code itself, not this extension. I'm guessing the extension is return some invalid object back to VS Code, which is then causing the exception.

If you'd like to try debugging this yourself, go for it! The extension is open source at this GitHub repo, and [the code that's logging the error message is here](https://github.com/microsoft/vscode-arduino/blob/faf77358cf8ccc436d56a57e4fc3891c133b0e49/src/serialmonitor/serialMonitor.ts#L166-L172). Something inside that `try` block is going wrong. The [development instructions in the README](https://github.com/microsoft/vscode-arduino#development) should still be up to date, but let me know if you have any issues or would like help fixing this.

Triangle4, 2022-01-19:
I would definitely like help fixing it. I am not strong enough a developer to be working on vscode source. I will dig in though and see what I can figure out. 
I will let you know if I arrive at a conclusion.
Can I ask, does this happen to you, when you open the serial port? Otherwise I will try it on a few other computers and see if it is reproducible.

benmcmorran, 2022-01-19:
No worries! We welcome external contributions, but there's definitely no obligation to fix the bug yourself.

I haven't seen this particular issue before. If I can't figure anything else out, I'll try adding some additional logging in the next release of the extension so that we can better narrow down the issue you're having.

Triangle4, 2022-01-19:
So I tried it on one of my other machines and it works. There is no input. 
I noticed that the VSCODE versions do not match.
After installing the lastest of vscode, this error is gone.

closed 2022-01-19 by Triangle4

benmcmorran, 2022-01-19:
Glad you got this figured out! It's possible to send commands to the serial port by using commands. On Windows type `CTRL + SHIFT + P` and search for "serial port".

[image]

Triangle4, 2022-01-19:
Thanks Ben. 
I am curious, Lets say I wanted to place a button next to the serial monitor button
Where would I start?

And if I am successful, where would I share that code so you can implement it in future extensions.
I sent you an email, if you are open to it we can discuss this there. I know you are busy so I won't take more than a minute of your time.

Thanks again for your help.